cleanflight/cleanflight #1720: Add support for binding R616XN receiver

sjtrny, 2016-01-24:
The R616XN receiver is a nano PPM receiver. To put it into bind mode you set or connect the signal pin to ground. However this is a huge pain if you have the R616XN direct soldered to the FC and buried in a small quad.

This issue is a feature request for adding a small CLI utility to assist with binding. It would be used as follows:

    set ppm_bind = 1

This will set a variable `ppm_bind` in the `master_t` struct which is stored on the EEPROM.

Next the user will reboot. Upon reboot the FC will check if `ppm_bind` is enabled. If so it will set the corresponding PPM pin for the target to GND. This has to happen fast.

Once the bind process is complete the user will then set

    set ppm_bind = 0

Which sets the variable to 0 and normal operation is resumed.

ledvinap, 2016-01-24:
There is already spectrum bind code that handles similar problem ...

sjtrny, 2016-01-24:
Yes I am well aware. However the binding procedure is very different and the pin assignment is also different.

ledvinap, 2016-01-24:
The mechanism to configure/reset binding and alter pin soon enough is the same. So new binding procedure may be modeled after it

hydra, 2016-01-27:
agreed

closed 2017-02-25 by hydra