cleanflight/cleanflight #1722: ARM in Mode tab should give visible danger warning

timdonovanuk, 2016-01-25:
I noticed this in EZ-GUI (android app) today, when setting up modes, and confirmed it in Cleanflight.

Incorrectly/accidentally setting ARM mode to a aux channel and range that's 'on', will instantly cause props to spin (when MOTOR_STOP is disabled, or even when its enabled, if throttle is up)..obviously this is because the quad has been armed. And this is correct however..

**All other actions in Cleanflight that might cause a spin up come with a warning** (e.g. the motor page, setting oneshot etc.). This is even more dangerous, as it's not just arming and spinning at low-idle speeds (as with normal stick arming where throttle is naturally at zero), but you're now at the mercy of wherever the sticks of your transmitter are.

Reporting this as I had it happen to me in the field when (miss)using EZ-GUI. Obviously on the bench using Cleanflight I have props off, but not everyone follows this advice, and there is a danger here.

Not sure the best way to prevent this - perhaps do not allow the user to save the Modes page if a save will result in ARM on.

readerror67, 2016-01-25:
Don't see how this would make things any safer, just don't have the battery hooked up when doing risky things.

timdonovanuk, 2016-01-25:
I completely agree you should not have the battery hooked up when doing risky things, and this is not the place for discussing best practices I already agree with. However, poor and momentary lapse of judgement do occur. 

Generally Cleanflight provides at least a cursory warning to a user when their actions are potentially dangerous, in the examples I provided. This action is arguably more dangerous and yet can be completely innocently triggered.

joshuabardwell, 2016-01-26:
Isn't this actually a configurator issue, not a Cleanflight issue? Configurator is its own repo.

SteveAmor, 2016-01-26:
@timdonovanuk could you close this issue and raise it in the Configurator repo.  Thanks.

closed 2016-01-27 by hydra

timdonovanuk, 2016-02-05:
Yep, will do. Apologies for this.