cleanflight/cleanflight #1725: Naze32 hangs right after arming when MSP telemetry is enabled on UART1

gkvirkvia, 2016-01-27:
As the title states. Having MSP telemetry on UART2, or non-MSP telemetry on UART1 is fine, motors spin up, drone is happy.
However, when MSP telemetry is enabled on UART1, upon arming motors spin up just briefly(100 ms?) and stop. The board becomes unresponsive after that and stays like that before I cut the power.

The behavior is present in Cleanflight 1.10, 1.11 and laters Betaflight.
The only settings I changed from their defaults are UART1 config obviously, feature TELEMETRY and RX_PPM and ONESHOT.

Hope this info is enough.

joshuabardwell, 2016-01-27:
I think that MSP telemetry has been removed for 1.12.

https://github.com/cleanflight/cleanflight/commit/1e098bf1990feab5dc4987526a9557b6e54fc28c

gkvirkvia, 2016-01-27:
Hm. Does that mean that Cleanflight is dropping support for MWOSD?

lostcontrol, 2016-01-27:
@gkvirkvia No, MWOSD still works, it uses MSP, not MSP telemetry.

gkvirkvia, 2016-01-27:
Ah I see!

closed 2016-01-27 by gkvirkvia

joshuabardwell, 2016-01-27:
MSP = two-way protocol. MSP telemetry = one-way telemetry using the same protocol format. Basically, Hydra's argument is that mostly nobody uses MSP telemetry, and those who do would be better off to use regular MSP instead.

joshuabardwell, 2016-01-27:
MSP itself is not going anywhere. In fact, if you look around here on Github, there has been some discussion of removing the CLI entirely, and replacing it with a GUI front-end that uses MSP to communicate with the board. Every CLI command has an MSP equivalent, basically. This would free up a LOT of program memory, since text parsing is pretty memory intensive.

gkvirkvia, 2016-01-27:
Thanks for giving the context. To your previous comment, it might be that there's unpredicted interaction between MSP telemetry and MPS code. I'll try to verify that once I get back home by enabling MSP and telemetry on other UARTs.

Also just being curious here: I assume MSP telemetry just rotates telemetry packets in a loop as it's TX only. Does MSP (two way) do the same by default, or it's being actively polled by telemetry device(MinimOSD in my case)?

joshuabardwell, 2016-01-27:
I don't know the answer to your question. I believe that MSP telemetry is one-way, but I don't know the details.

gkvirkvia, 2016-02-02:
Just in case anyone stumbles upon this thread, dug up MWOSD source up, it indeed is actively polling MSP protocol for stats. Keeping UART1 just in MSP mode worked fine with MWOSD.