cleanflight/cleanflight #1739: Missing RPM sensor using SBUS/S.port?

joeynuggetz, 2016-01-30:
Upgraded from D4R-ii to X4R and I do not see any RPM sensors available. Running OpenTX 2.1 on Taranis X9D Plus.

Also, tmp1 sensor shows erratic values using X4R. Normal temperatures when using CPPM/D4R-ii.

Cleanflight 1.11.0  Target =RMDO

mosi0815, 2016-01-30:
Temp1 and Temp2 are used for flags. Like flight mode, arm status etc. That's why you see weird values jumping around.

joeynuggetz, 2016-01-31:
I dont think that's true. While running the D4R I took a heat gun and gave the PCB a blast of hot air and the temperature reading went up. 

What about RPM?

joshuabardwell, 2016-01-31:
@joeynuggetz are you suggesting that the D4R2 has a temperature sensor built in?

mosi0815, 2016-01-31:
The smartport telemetry code is entirely different to the frsky telemetry code. Frsky sends the temperature of the gyro chip or the temperature of the baro chip depending on which one is available. Thats temp1. Temp2 represents the GPS quality.
In smartport mode temp1 and temp2 are both filled with flags. Try it. arm your copter and look what happens to temp1.
The smartport code doesn't send RPM at all. Frsky code transmitts the battery capacity if the craft is disarmed else the throttle position.

joeynuggetz, 2016-02-04:
For sure. That's what I was asking. D4R (Frsky telemetry) sent temps and noticed they weren't there using S.port. Was just curious if it was a protocol thing or something that was missed. Thanks for clearing it up. So are these "flags" useful? Why are they being sent? 

Regarding RPM, what do you mean about throttle position? The numbers I was getting on Frsky telem when using RPM appeared to be legitimate RPM values unless I was misinterpreting them. Certainly didn't look low enough to be throttle positions.

closed 2017-02-25 by hydra