cleanflight/cleanflight #1740: New Makefile behaviour flashes wrong target (regression)

stronnag, 2016-01-30:
The new behaviour of #1724 has an unfortunate side effect of flashing the wrong firmware. 
Prior to #1724:
```
make TARGET=CC3D  # builds the CC3D hex
make TARGET=CC3D flash  # flashes the CC3D hex onto the FC
```
now:
```
make CC3D  # builds the CC3D hex
make CC3D flash  # flashes the CC3D hex onto the FC, then builds the NAZE hex, and flashes that
 ```
If this is intended behaviour it needs to be documented in the release notes; if not please revert or revisit.

[session_log.txt](https://github.com/cleanflight/cleanflight/files/110715/session_log.txt)

ledvinap, 2016-01-30:
I'll look into it. `make TARGET=CC3D flash` should work correcly

ledvinap, 2016-01-30:
@hydra: I can't find any simple fix, here are possibilities I can think of:
-  replace target-specific rules with dummy ones using `if` on MAKECMDGOALS. The conditional section will be quite large (almost all of makefile), Makefile is already complicated enough
- Rename current Makefile as target-specific (Makefile-target) and create new Makefile that will handle TARGET (by invoking Makefile-target) and global goals. This is IMO cleanest way (target specific makefile will be slightly simpler / stucture is quite clean). It is in similar to first option without huge `if`.
Only disadvantage is that this change isn't 'small'
- Set TARGET to first target in MAKECMDGOALS, possibly generate dummy rule for TARGET. Ugly, but extent of changes will be quite limited ... 
- accept the limitation, use TARGET= for non-compilation goals, accept the fact that 'make SPARKY binary' builds NAZE too

closed 2017-02-25 by hydra