cleanflight/cleanflight #1741: Sbus not working in cleanflight

Propitiation, 2016-01-30:
serial_RX option will not save under UART2. I have tried everything, it just won't save, I flashed baseflight on my quad and took it for a flight, so i know my wiring is correct.

joshuabardwell, 2016-01-31:
This is a support question. Github is for bugs and other development issues. Please ask over on the RCGroups thread and close this issue.

http://www.rcgroups.com/forums/showthread.php?t=2249574

Propitiation, 2016-01-31:
actually this is not a support question, this is a bug issue. I know its a bug, because my quad flies perfectly in baseflight without changing a single wire. Cleanflight is not operating properly, if there is a workaround great! But that's exactly what it will be... because this is a bug.

joshuabardwell, 2016-01-31:
It's not a bug. Most likely, what has happened is that you are trying to enable serial RX on the UART before you have set your receiver type to serial and selected the correct serial provider in the Configuration page. You would have found this out if you had done what the "new issue" readme told you to do, and gone to the support forum, instead of creating a new issue.

So go give this a try:

1. go to Configuration and set RX_Serial as your receiver mode.
2. Set your Serial Receiver Provider as appropriate for the type of receiver you have.
3. Click "Save and Reboot".
4. Go to the Ports tab and tick "Serial RX" on the UART you want to use. MAKE SURE NOTHING ELSE IS ACTIVE ON THAT UART.
5. Click "Save and Reboot"

et voila.

Let me know if this fixed your issue.

Propitiation, 2016-01-31:
Ok I actually just got a reply from a nice guy and now I know what was going on. In hindsight, it may not be considered a bug, but the gui could have been a tad bit more intuitive about it. Under UART2 all I needed to turn on was Serial RX, while leaving everything else as is. Turning on MRP will deactivate Serial RX and MRP. Sorry for freaking out. Looking forward to trying cleanflight for the first time.

closed 2016-01-31 by Propitiation

joshuabardwell, 2016-01-31:
Glad your issue is solved.

joshuabardwell, 2016-01-31:
PS: I'm glad that @thenickdude was able to help you, but Github is still the wrong place for this type of question. Please post on RCGroups next time, and you will get good help without taking the attention of the devs.