cleanflight/cleanflight #1745: Is softserial1 on IO1?

johnlenfr, 2016-01-31:
Don't know if it is a problem in GUI or in the SP Racing F3 hex file but to have Telemetry on Softserial 1 I have to select Hott telemetry on Softserial 2 in the GUI.

Is there someting inverted in the code or I missed something in my hardware?

Thought config was like this:
PIN-IO1-IO2
5-SS1Rx-SS2Rx
6-SS1Tx-SS2Tx

Here is my setup:
Running - OS: Windows, Chrome: 48.0.2564.97, Configurator: 1.1.0
Flight controller info, identifier: CLFL, version: 1.11.0
2016-01-31 @ 09:40:21 -- Running firmware released on: Nov 28 2015 15:29:46
2016-01-31 @ 09:40:21 -- Board: SRF3, version: 0

Thanks

hydra, 2016-02-04:
please provide configuration dump and wiring photo if possible

johnlenfr, 2016-05-17:
Here you can find attach print screen of my config (since my first post I change to new CF v1.12).
[image]

You can see that I have to select Hott Telemetry on SOFTSERIAL 2 but thing is I cable my telemetry on IO_1 port as you can see on the photos.
[image]

[image]

So is there a mismatch in GUI or Hex file?

Thanks

closed 2017-02-25 by hydra