cleanflight/cleanflight #1748: Random cut outs on CC3D

edusteinhorst, 2016-02-01:
I couldn't find an issue for this on CF, but it exists on borisbstyle/betaflight#64.

Tested on a CC3D with CR 1.12-RC1.

I get cut outs randomly, where all motors cut at the same time, usually under 2 minutes after arming. The motors cut out, the blue led turns off and the yellow led comes on. If I lower and raise throttle it rearms (blue led) and motors come back on.

This only happens when gyro_sync=on. Also, the problem seems to be eliminated or drastically reduced when gyro_sync_denom=2 (couldn't reproduce it). Switching blackbox on or off had to effect.

The issue doesn't seem to be related to ESCs, as it happens even without ESCs plugged.

The disarm happens because of periodic glitches on the aux channels, as seen below. The disarm doesn't happen if a switch is not used for arming.

[image]

Also tried setting rc_smoothing=off, but it had no effect.

Any ideas?

borisbstyle, 2016-02-01:
The problem is the oneshot timer clash on PPM.

Disable oneshot125 and you will be fine till its solved

hydra, 2016-02-02:
@borisbstyle can you give more details into the cause?

Additionally the 'channel stability' PR #1710 I created might help with this (but agreed it masks any underlying problem).

edusteinhorst, 2016-02-02:
@hydra

Checkout the issue borisbstyle/betaflight#127

I'm not familiar with the project, but I'll try giving it a go.

borisbstyle, 2016-02-02:
@henn1001
Can explain this better.

I even don't have CC3D anymore to troubleshoot this. From my understanding there is a timer clash between PPM and oneshot125 timer. Openpilot already moved PPM pin to somewhere else.

@hydra 
This is exactly why I wanted more disarming security. The code as it is now needs 1 single glitch to fall out of sky.

edusteinhorst, 2016-02-02:
Just an idea, but it might be worth having some hysteresis on the disarm trigger, be it switch or sticks.

henn1001, 2016-02-02:
Yeah it would be cool if @hydra could look into borisbstyle/betaflight#127 :) 

Just for testing https://github.com/henn1001/cleanflight/commit/66edf2434cf496896ef9f8f7cf463cd40efd4f20

Compiled from latest source
[cleanflight_CC3D.zip](https://github.com/cleanflight/cleanflight/files/113925/cleanflight_CC3D.zip)

Give it a go on the bench without props and see if you can reproduce it - PPM goes now to the last pin (8 i think - where RSSI was previously)

edusteinhorst, 2016-02-02:
@henn1001

Awesome, I'll test it tonight. Is there any other pin the RSSI analog input can be moved to?

edusteinhorst, 2016-02-02:
@henn1001

Ok, I tested it. The PPM input did change to the RSSI pin, but it only works with OneShot disabled. If I enable it I get not PPM signal, all channels completely still. Is OneShot messing with the TIM2 config?

henn1001, 2016-02-02:
Welp

edusteinhorst, 2016-02-04:
I'm working on this, just to let you guys know.

edusteinhorst, 2016-02-08:
@henn1001 @hydra @borisbstyle

This patch wasn't working because TIM2 channel 3 was mapped as a motor output (PWM12/S6_OUT), so whenever Oneshot was enabled it would change the prescaler for the entire TIM2 to increase resolution (can't change prescaler per channel).

The way OneShot is implemented now forcefully overflows the timer to have an imediate output when motors are set. To allow it to coexist with PPM, the hack first backs up the current counter before forcing the overflow (by seting the UG bit in the TIMx_EGR register), and later restores it in the interrupt routine so the PPM algorithm doesn't freak out. The problem here is probably a race condition when doing that. 

Please see edusteinhorst@930c6ce. I also opened a PR #1790.

derFrickler, 2016-04-29:
I get exactly these cut outs described in the first post now with CF 1.12 where CF 1.11 worked fine with the same setup. CC3D Nano, CPPM RX with arming on AUX1=HIGH, OneShot activated. 
Problem appears quite often directly after lift-off, especially in LEVEL mode (AUX2=LOW) where Horizon and RATE seem to work fine. Flashing back to 1.11 seems to fix the issue. 
Got 2 cut outs on video when i did the some tests. after switching off LEVEL it worked fine.
https://www.youtube.com/watch?v=bvMQd1UvFio&feature=youtu.be

edusteinhorst, 2016-04-29:
Hi,

I submitted a PR for this, #1790 . It's scheduled for 1.13.

I haven't checked if the issue wasn't there in 1.11, and the glitch should happen independent of flight mode, so I can't confirm you have the same issue. This issue is related to OneShot. if you disable OneShot it won't happen.

If you want to test the fix, there's a zip file with CF compiled for CC3D in #1790 . Just beware that it's based on the code base from early feb.

derFrickler, 2016-05-01:
Thanks, 
As i don't have the issue on older CF Versions or the current BF onmy other quads, it seems to have come in with the latest releases, switching back to 1.11 worked for me so far. Think i will change to BF on that machine too then.

closed 2016-05-06 by hydra