cleanflight/cleanflight #1750: GTUNE not defined for CC3D\target.h

paulwas, 2016-02-01:
Is there a reason that GTUNE is not included in CC3D?

I loaded the latest code (1.12.0) onto a CC3D and noticed that there was no GTUNE mode. So I looked at the code and noticed that "#define GTUNE" was not included in "main\target\CC3D\target.h". So I added it to my copy and built it then loaded it onto my CC3D and now it's there. I've not had a chance to test it out in a quad chopper yet so I don't know for sure that everything is correct. Thanks for any info..

joshuabardwell, 2016-02-01:
I believe that it's removed to make room for the OpenPilot bootloader.

paulwas, 2016-02-01:
Thanks. If that's the case I can deal with it locally.

borisbstyle, 2016-02-02:
CC3D even without bootloader is near the end of its flash.

Cleanflight doesnt support bootloader anymore btw

artikot, 2016-02-25:
can you paulwas or anyone else share a cc3d.hex with gtune please?

paulwas, 2016-02-25:
Hey artikot,
I sync'd and built a version enabling gtune and I've put it up on dropbox (link below).
However, I've not had a chance to make sure it fits or works so just be aware of that and let me know what you find. Good luck..

[https://www.dropbox.com/s/didm6ypgbax97o6/v1.12.0-XXX%20cleanflight_CC3D.hex?dl=0](url)

artikot, 2016-02-25:
Hi paulwas,
Just flashed it and the gtune is there but the air mode is gone (doesn't matter as I'm not using it yet).
Also noticed default looptime is 3500. I think that before it was around 1000.
What is the correct looptime for littlebee esc with blheli and oneshot125?

paulwas, 2016-02-25:
When I get a chance I'll see if I can figure out what's going on with the air mode.
And the default looptime across the board is 3500. I've not used the oneshot esc's but I seem to remember that the recommend looptime for them was 2500 (400/sec) so you might try that.

artikot, 2016-02-25:
Thanks alot for firmware and suggestions.

sreichholf, 2016-05-11:
Well maybe I can give a short statement here...
Essentially using the current master GTUNE won't fit into the CC3D's flash without removing some other feature. This is most probably the reason why it's missing in the target.h

I've built myself a version without Beeper-Code (I don't have the HW-Mod so it won't work here anyways) but WITH GTune enabled. 
Unfortunately I haven't been able to test it because of very windy weather conditions here.

There actually are other open issues (like #1621 and #922) related to "getting more space for flight features".

Until then I don't see much of a sultion other than "custom builds".
Considering that GTUNE is kind of a "one-time-use" feature it's probably not preferrable to kick out every-day-in-flight stuff for it.

The only other way would be providing multiple builds for 128KB F1 Targets.

closed 2017-02-25 by hydra