cleanflight/cleanflight #1753: Corrupt CLI output when using UART ports with latest 1.12.0-RC2

hydra, 2016-02-02:
As noted by @borisbstyle there are issues on non-VCP targets.  MSP is fine but the cli output is corrupt.  

Related to PR #1143, PR #1151

borisbstyle, 2016-02-02:
What about check in cliprintf function?

When VCP connected just use printf and otherwise use buffer?

hydra, 2016-02-02:
fix confirmed by @stronnag

closed 2016-02-02 by hydra

borisbstyle, 2016-02-03:
@hydra 
It does work now but is still not 100%. I get echos on commands now

borisbstyle, 2016-02-03:
when typing status you get this
# statusstatus

stronnag, 2016-02-03:
Don't see that here:

````
# feature
Enabled: VBAT FAILSAFE RX_PARALLEL_PWM 

# version
# Cleanflight/NAZE 1.12.0 Feb  2 2016 / 17:50:07 (1249724)
# status
System Uptime: 66 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.02
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Cycle Time: 3513, I2C Errors: 0, config size: 2056
````
I'll try the later commit next.

borisbstyle, 2016-02-04:
yeah it seems good. It was my mistake on the last pr.

Only thing I notice now is that it didnt like large paste dumps. I believe I tried naze board without VCP. Will test some VCP boards later as well.