cleanflight/cleanflight #1758: Passthrough mode missing from custom airplane mixer - Nase23 rev5

badfrogg, 2016-02-03:
Nase23 rev5
I noticed that if you select airplane mixer, you will have passthrough mode on the modes tab. But, if you select custom airplane, the mode "Passthrough" disappears from the modes list. Is there a command in CLI to enable this pass-through mode

And just to be clear i mean the mode "passthrough" for servos not serial passthrough.

Another odd thing is if you go back to regular airplane mixer the passthrough does not come back.

hydra, 2016-02-04:
Please test new release candidates ASAP.  Raise a new bug and reference this if it's not OK.

closed 2016-02-04 by hydra

badfrogg, 2016-02-07:
Yes, The 1.12.0 RC2 has fixed the issue (enhancement). Passthrough Mode now avalable in Custom Airplane Mixer.

Thanks!

hydra, 2016-02-07:
Great, thanks for the feedback and bug report.

commit 25c7db49 (2016-02-04): Allow PASSTHRU mode when using custom airplane mixer. fixes #1758