cleanflight/cleanflight #1759: SPRacingF3 ESC/Servo Configuration document

Ravid824, 2016-02-03:
Can someone point to any documentation describes the SPRacingF3 ESC/Servo port configuration ?
Previously I was using Naze32 that has 6 outputs while 4 were used for the motors and two for gimbals servos.
The SPRacingF3 has 8 outputs and I couldn't find anywhere explaining where to connect the servos and what the is the use the extra two outputs when I use QUAD.

metalnow, 2016-02-03:
did you check the official website?

Ravid824, 2016-02-03:
Yes, nothing besides
"16 PWM I/O lines for ESCs, Servos and legacy receivers. 8 available on standard pin headers. 8 via side mounted connectors"

Ravid824, 2016-02-10:
I connected a scope and found out the gimbals pins:
ESC/Servo output Pin7 -  Pitch (controlled from CF by servo 0)
ESC/Servo output Pin8 -  Roll (controlled from CF by servo 1)

I am not sure what Pin 6 & 7 do  with Quad configuration (looks like that they are configured and  emitting motor signal). Do they these pins has any other use beside of motor connections?

Still it worth to update the documentation.
Thank you

closed 2017-02-25 by hydra