cleanflight/cleanflight #1762: Heading indicator changing by 1 degree every few seconds

joeynuggetz, 2016-02-04:
I'm on my 2nd RMRC Dodo F3 board and the replacement board is doing the same. If I perform a calibration and let the board sit perfectly still, the heading indicator and graphic in the CF setup page will increment by a single degree every 3 or 4 seconds. Graphic updates as well and if I let it sit it will spin by 1 degree forever.

I RMA'd my old board because of drifting issues in angle mode and weird issues with yaw which I attributed to a bad board and the concluded that the heading indicator showing that the board was rotating could someone be responsible for the unexpected performance with yaw while in level mode.  

Currently running on RMDO 1.12RC. 

This does NOT happen with my Beef Brushed Board. Neither board has a mag.

badfrogg, 2016-02-07:
I am seeing this too on naza32 rev5 1.12.0 RC2 (6 dof , no mag)

skaman82, 2016-02-07:
Having a similar issue with my scisky board. Roll offset is rising by itself until the quad drifts extreemly to the right. See: https://github.com/cleanflight/cleanflight/issues/1780

joeynuggetz, 2016-02-08:
I'm starting to think this is normal. I RMA'd a board that was experiencing this (board had other issues like only registering 60 degrees of movement in the setup screen when in reality I rotated my quad a full 360 deg). After I removed the old board and tested the new board - the old board stopped doing the heading drift thing and the NEW board was doing what the old board used to do. New board shows a 1 to 1 correlation between rotation and the heading and graphic displayed on screen unlike the old board but I wonder if ANY heading drift while the quad is still suggests that gyro health between boards are different? Who knows.

DeetAus, 2016-02-17:
I'm seeing it on a new SPracingF3mini Board

kronikabis, 2016-04-05:
when you look at your channels, does the particular channel have the numbers moving slightly/jittering? ie. 1000->1001->1000 to ad infinitum. This was the cause of my similar issues. Would yaw in flight to the left (on 2 cc3d's and 2 naze32's). setting a dead band of 5 on the channel sorted it for me. It didn't matter if it was still doing it in the gui, flight was perfect after that.

tedbarth, 2016-10-24:
I notice the same with the current build on  an F3 and with an old Version (sry don't know what version) in my Naze32.

closed 2017-02-25 by hydra