cleanflight/cleanflight #1780: Drift issue in ANGLE and HORIZON

skaman82, 2016-02-06:
My micro quad is drifting to the right in ANGLE and HORIZON mode. It gets stronger over time until I really have to fight the quad hard not to drift. As you see in this video the drift grows over time - the quad is standing still on the table the whole time. Tested with CF 1.12, 1.11 and Betaflight 2.3.5. Tested the Naze32 and AlienWiiF1 versions. I think this is a bug that needs fixing.

Video: https://youtu.be/PO5ohBe4a_E
Board: SciSky32. 
Dump: 

# version
# Cleanflight/ALIENWIIF1 1.12.0 Feb  5 2016 / 14:18:34 (360fd1a)
# dump master

# mixer
mixer QUADX
mmix reset
mmix 0 1.000 -0.414 1.000 -1.000
mmix 1 1.000 -0.414 -1.000 1.000
mmix 2 1.000 0.414 1.000 1.000
mmix 3 1.000 0.414 -1.000 -1.000
mmix 4 1.000 -1.000 -0.414 -1.000
mmix 5 1.000 1.000 -0.414 1.000
mmix 6 1.000 -1.000 0.414 1.000
mmix 7 1.000 1.000 0.414 -1.000
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 2000
set emf_avoidance = OFF
set i2c_highspeed = ON
set gyro_sync = ON
set gyro_sync_denom = 1
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = OFF
set input_filtering_mode = OFF
set min_throttle = 1000
set max_throttle = 2000
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 32000
set servo_pwm_rate = 50
set retarded_arm = OFF
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = NMEA
set gps_sbas_mode = AUTO
set gps_auto_config = ON
set gps_auto_baud = OFF
set serialrx_provider = SPEK2048
set sbus_inversion = ON
set spektrum_sat_bind = 5
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 188HZ
set gyro_soft_lpf =  60.000
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set yaw_control_direction = 1
set pid_at_min_throttle = ON
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set failsafe_delay = 2
set failsafe_off_delay = 0
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = 0
set rx_min_usec = 885
set rx_max_usec = 2115
set acc_hardware = 0
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 45 45 100 -1
servo 1 1000 2000 1500 45 45 100 -1
servo 2 1000 2000 1500 45 45 100 -1
servo 3 1000 2000 1500 45 45 100 -1
servo 4 1000 2000 1500 45 45 100 -1
servo 5 1000 2000 1500 45 45 100 -1
servo 6 1000 2000 1500 45 45 100 -1
servo 7 1000 2000 1500 45 45 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = ON
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = NORMAL
set acc_cut_hz = 15
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_delta_method = MEASUREMENT
set pid_controller = NORMAL
set p_pitch = 36
set i_pitch = 30
set d_pitch = 23
set p_roll = 36
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  1.400
set i_pitchf =  0.400
set d_pitchf =  0.030
set p_rollf =  1.400
set i_rollf =  0.400
set d_rollf =  0.030
set p_yawf =  3.500
set i_yawf =  0.400
set d_yawf =  0.010
set level_horizon =  3.000
set level_angle =  5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500
set dterm_cut_hz =  0.000
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 130
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 100
set tpa_rate = 0
set tpa_breakpoint = 1500

#

joeynuggetz, 2016-02-07:
So you're saying that it drifts when running all those different versions you listed? Sounds like a hardware problem. Make sure your trims are centered. I've seen this happen before but in Acro and was caused by my trims being non-centered.

skaman82, 2016-02-07:
All trims are centered, all channels are at 1500. 
This even happens when TX is off an the model is connected to CF and lying still on the table (see video).

joeynuggetz, 2016-02-07:
I see. I was having an issue with the yaw drifting. Feedback I was getting is that its normal for the heading to drift when you dont have a mag. 

To be honest, if you load different firmware on that board and it happens all the time then it has to be the board, no? 

There was a thread on here talking about drift some time ago but I think it was closed.

digitalentity, 2016-02-07:
It's not necessarily the board itself. If the issue is the same on
different boards and different firmwares it is likely not a firmware issue
but something else. Do you have this issue with stopped motors add well?

skaman82, 2016-02-07:
Thanks. I just ordered a spare FC. Hopefully it arrives in the next three weeks. Will check if same issue occures and close this if not.

skaman82, 2016-02-07:
@digitalentity yes, this happens also with motors of and lying flat in the table. See video in the first post.

digitalentity, 2016-02-07:
@skaman82 
Can you record a video of "Sensor" tab with Gyro and Accel graphs? I have an idea but I'd love to confirm it first.

ledvinap, 2016-02-07:
@skaman82 : cli `status` may help too

skaman82, 2016-02-07:
# status
System Uptime: 34 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.07
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Cycle Time: 1001, I2C Errors: 0, config size: 2020

skaman82, 2016-02-07:
@digitalentity I've made a new video: https://www.youtube.com/watch?v=5alUJJxPxdY
At the end of the video I've tried to do an new calibration, but it didn't work.

I was able to investigate this further. This only happens if I set "set motor_pwm_rate = 16000" for my brushed motor setup in the CLI. With the defaults I wasn't able to reproduce this.

digitalentity, 2016-02-07:
Ok, this must be interference issue from motor pwm. Pwm frequency is higher
than gyro sampling frequency and this is probably causing aliasing.

set imu_dcm_ki = 50 will solve level drift for you, but in rate mode you
still will be having issues with drift.

skaman82, 2016-02-07:
yes, set imu_dcm_ki = 50 has done the trick. The drift is still there, but now it stops at 0.4 instead of 3.0.

skaman82, 2016-02-07:
@digitalentity had further testing with imu_dcm_ki=50. The quad is not drifting to the right anymore but to the left with the same behavior as before :(

skaman82, 2016-02-07:
Increasing looptime to 1600 helps.

digitalentity, 2016-02-07:
@skaman82 another confirmation of aliasing - you changed sampling frequency and this moved aliased noise from motors from DC bias (0hz) to some other frequency.

robschii, 2016-02-10:
i have the same problem. i m using the cc3d witch the last 1.12.0rc2. the drift is to the left and back. i have many times recalibration the acc but with no success.  some minutes ago i testet the copter with librepilot and there is no drift.  when i disarmed and armed again, the drift is away but with flight time it comes back

lostcontrol, 2016-02-10:
I once add a bad MPU-6xxx on a Chinese Pixhawk-clone. I could see from time to time the AHRS drifting a lot (really a lot) while the copter was laying flat on the bench.

This was with APM:Copter but maybe something similar could also happen with Cleanflight/Naze32/CC3D/etc!?

I managed to switch the MPU-6xxx on this board with a new one and... tada... problem solved. The chip was most likely defective, yet not enough to not function at all, but bad enough to cause fly-aways from time to time.

Here is the link to the thread if you are interested: http://ardupilot.com/forum/viewtopic.php?f=100&t=12091#p32999

robschii, 2016-02-10:
so i switch back to 1.11.0  and the drift is away.  seems to be a bug in 1.12.0rc2??

lostcontrol, 2016-02-12:
@digitalentity Don't you think this can be caused by a too high vibration level?

skaman82, 2016-02-12:
@lostcontrol I don't think so, because: 

1. this happens in the configurator even when the model is sitting still on the table with motors off (heading  "roll" changes in the setup tab, see video in the first post)
2. the drift seems only to occur when PWM is et to a high frequency like 16000 or 32000 or 4000

robschii, 2016-02-12:
the problem are not the vibrations, so why it is not in earlier versions and only in the last rc2?  i use s.bus

joeynuggetz, 2016-02-14:
I noticed today that with the latest version of clean flight 1.12rc2 that the quad copter would drift to the right and back. I would land and recalibrate my gyros and lift off and the copter appeared to level correctly but then the drift reappeared after stick inputs. I'm going to try adding some dead band to my pitch and roll to see what happens and rule out my taranis. Dodo rev 3

lostcontrol, 2016-02-14:
I also notice this on a small RotorX Atom this week-end. The AHRS is rock stable on the bench. Acro mode is fine but angle (and horizon) show drift. I tried to compensate the level while doing an accelerometer calibration but the drift reappears quickly after take-off. I tried acc_cut_hz = 60 and imu_dcm_ki = 50 but it doesn't help much. I still strongly suspect this is all related to vibrations (at least in my case). This was with 1.12.0-RC2, I should try with 1.11.0 to see if it's better there.

ledvinap, 2016-02-14:
blackbox log with this problem will be great ..

digitalentity, 2016-02-14:
Neither imu_dcm_ki not acc_cut_hz will help with frequency aliasing which
is likely your case.
Is your FC hard mounted to the frame or do you use some sort of vibration
damping?
A log indeed will be helpful.

skaman82, 2016-02-14:
My scisky is mounted with thick gyro tape to the frame. Scisky has no flash, so I can't provide a log :(

lostcontrol, 2016-02-15:
@digitalentity It is hard mounted on the frame (with nylon standoffs). Unfortunately, no blackbox (I have an openlog lying around, I should give it a try). I moved from some foam mounting to nylon standoffs on another quad (ZMR250) but didn't flight it yet (no time, bad weather). This one has some resonances at mid-throttle, I'm curious to see how Cleanflight will handle this. Likely I will need to find dampening solutions (or balance my motors).

JoeB99, 2016-02-16:
Just flashed 1.12 RC2, and the drift is about 1 deg per 5 seconds. With 1.11 it is about 1 degree per 10 seconds. To reproduce on the desk, not in a quad: Flash the board, power-cycle, connect to configurator, calibrate acc. sensor.
My board is in a case, stuck down to the table.

JoeB99, 2016-02-16:
Just uploaded 1.10 and there is no drift at all.

IcyLemur276, 2016-02-16:
I have the Colibri with my TBS Powercube and I too have noticed Drift issues in either Angle or Horizon mode.  Rate is just fine.  I started off with Cleanflight 1.11 and it had heavy drift, I then tried out Betaflight 2.3.4 and the drift was much less but still there and nothing a little bit of accel trim wouldn't fix. Tried with 2.4.0 the drift issue returned.  Maybe I'll give 1.10 a try, but I really do not like to revert to older code and would like to see the latest code working fine.

joeynuggetz, 2016-02-17:
Did you guys know that there's a special version of blackbox that allows
you to graph the amount of time the accelerometer is used for attitude
calculations vs the gyros? When I loaded up a log file it showed that the
attitude info on my quad was mostly being calculated from the gyros which
is a problem. I understand this to be due to the amount of high frequency
noise. I'm in the process of trying to isolate any vibrations from hitting
the FC. I currently have kyosho zeal gel in use but even with that the
drifting persists. Looking to concoct a suspension type system for the FC.

DeetAus, 2016-02-17:
I did a maiden on a new airframe yesterday with an SPracingF3Mini, 1.12.0rc2, S-bus input from Futraba RX. On the bench all centres are 1500

It will take off fine but after a few seconds it tilts and drifts to the right. Land and re-do ACC_CAL via sticks and its OK again for a few seconds then it tilts to the right again

The board is hard mounted on nylon spacers.

I went back to the bench and balanced motors and props thinking it may be vibrations. Haven't flown it again yet

May or may not be relevant but I do get the yaw drift while its on the bench
I don't have a Micro SD card handy sorry

joeynuggetz, 2016-02-17:
I heard that the yaw drift in normal. I'd be interested to hear how your
drifting to the right turns out. I tried zeal gel and now moon gel and I'm
ready to put it back the way it was.  On spacers with nuts holding it to
the bottom pdb plate. I even tried going back to 1.10 and it still drifts.
I'm going to install my CC3D if taking the moongel out doesn't make it
better. There's an older thread here about this issue with lots of
technical info and it seems the conclusion is that this is caused by
vibrations. Not sure what the fix is. I was going to attempt an oring
suspension mount setup but I don't have enough room on my quad to pull that
off.

joeynuggetz, 2016-02-17:
By the way did you perform a gyro or accelerometer calibration after you
landed? Gyro calibration makes more sense. No need no calibrate
accelerometers again if you already did that as the board shouldn't move
around in the frame. Gyro is left stick down and left and right stick down.

DeetAus, 2016-02-17:
Once it started drifting Gyro cal did nothing
only an ACC_CAL would fix it

as to the yaw drift, its REALY fast when first connected to config, but slows once the board has finished the start-up gyro cal

The inflight drift was on the very first test hover, I haven’t flown a second lipo on it yet

I have the old SPracingF3 running on v1.11 without this issue
I have about a dozen Naze32 boards around that don’t show the issue on older firmware

What I am seeing is NOT a vibration issue.

Regards,

David Traino

JoeB99, 2016-02-17:
You don't need to fly to see the drift, It happens sat perfectly still on the bench connected to the laptop.

DeetAus, 2016-02-17:
I can’t see the acc drift on the bench, only the yaw drift

Regards,

David Traino

joeynuggetz, 2016-02-18:
same here. Haven't noticed roll drift on the desktop tho I will take a look to make sure that's not happening.

garthberry, 2016-02-20:
I have noticed this problem too, quad sitting on the table plugged into the configurator. 

I first noticed this problem running 1.11.0 on a Super-X with "MX NANO OVERSKY 32 type X" FC. I had both roll and yaw drift (to the right and back). I have since moved over to betaflight where at first I would only see roll drift to the right. However, the last few 2.4.0 RCs I now see yaw drift, both roll and pitch are rock solid.

I don't usually fly the quad in horizon mode, however, if I turn it on while flying the quad will correct by the total drift up to that point in time. When using cleanflight this would typically result in the quad flipping if activated more than about 2 minutes after the start of the flight.

DeetAus, 2016-02-20:
Some updates
Initially I was testing the MW2.3(rewrite), So I also tested the normal MW2.3 as well
This is after making sure all motors and props are balanced, so vibration should be minimised

I get the same drift in both versions, but it seems less harsh with MW2.3

I DO NOT get yaw drift on screen with MW2.3

I have come back from the filed, don nothing else but plug into the computer and connected to config

What I see on the setup screen is the airframe tilted to the right, and in the top left corner it tells me roll is 3.1 degrees

Calibrating Acc returns it to level

In flight what I found was the drift would slowly appear then increase over the first minute of flight, then the angle sense to be stable for the remains of the flight

DeetAus, 2016-02-20:
Just did some bench tests

1. Left connected to Config, didn't move it, no radio. Sat for 6 minutes, nothing drifted
2. Turned on radio and stirred sticks for 3 minutes, nothing drifted
3. Ran motors (props off, via radio), ran for 3 minutes, stirred sticks. didn't move the airframe. At full throttle would get 1 degree nose down drift, it would return to level as soon as throttle dropped back. 
There was no roll drift

all the above was with the frame sitting on a notepad on the bench, not moving

4. Power to motors disconnected. Radio off. Violently rolled and pitched the frame for 2 minutes by hand. No drift. the frame shows level when placed back on bench

So I cant replicate the fault on the bench

DeetAus, 2016-02-21:
Just tested with the stable V1.12 and its the same

IcyLemur276, 2016-02-21:
Can you try 1.9?

DeetAus, 2016-02-21:
i can’t try 1.9 with this board (SPRacingF3mini) as the only code for the board is v1.12
I have v1.11 running on the oder SPRacingF3 and it has no drift
I Have 1.8 and 1.10 running on Naze32 boards with no issue

>

gregm412, 2016-02-23:
I am also having a drift back and to the left sp pro f3 running cleanflight 1.12. Can not find the LPF to turn it on its missing. Has anyone figured this out yet?

gregm412, 2016-02-24:
[blackbox_log_2016-02-23_195143 drifting.TXT](https://github.com/cleanflight/cleanflight/files/143768/blackbox_log_2016-02-23_195143.drifting.TXT)
 this is my black box log it was a bit windy but still drifting back even when wind was pushing it forward. can someone please help?

ledvinap, 2016-02-24:
@gregm412 : There is over 1g of noise in accelerometer reading. You may try adjusting filters, but it won't help. You will have to balance  your props better.

[image]

gregm412, 2016-02-24:
@ledvinap thank you so much for looking into this for me. I just bought some gel mounting tape and a better mount and a balancer based on your recommendation. Thank you again for your help

ledvinap, 2016-02-24:
@gregm412 : You should start with balancing. Mounting will mostly only mask the problem (it is mechanical low-pass filter. It will attenuate vibrations but also introduce delay)

gregm412, 2016-02-24:
@ledvinap I will for sure start with the balance first. I can't thank you enough I was about to get rid of this sp pro f3.

ledvinap, 2016-03-12:
@gregm412 : see #1946. Acc sampling frequency was changed in 1.12, making acc vibration problems much worse ..

gregm412, 2016-03-13:
OK so is there a fix for it?  If so how do I install the fix?  Thank you
for thinking of me.

lit1088, 2016-03-21:
my hardware is naze32, In angle mode, before fly ,I have do accelerometer calibration, in few minutes ,the roll and pitch will drift. what is matter? I've been testing it for a week, and it's all like that.

manors, 2016-04-11:
i have the same problem (in v1.12.0-RC2 and latter). 

when i set acc_cut_hz=4, there is no drift anymore. issues solved.

DeetAus, 2016-04-11:
I will test that this weekend

CerealKiller1, 2016-04-15:
Hello I am having the same exact issue on CF1.12 on a naze32 tricopter integrated board. It was fine on 1.9 threw 1.11 and also triflight but I tried to upgrade to 1.12 and this problem reared it's very ugly head. Always drifts to the right and back. I land and do a stick ACC cal and it's fine for maybe 2 minutes and right back to it so bad I have to give it half way left stick to hold position (channel output monitor shows I'm commanding AIL at -52 to not be drifting right).

Sorry no Blackbox :(

I will try the above suggestion if set acc_cut_hz=4 first thing tomorrow and report back. Note I don't understand what imu_dcm_ki is buy I did try playing with it, from min to max value and it made zero change in the issue. As above the ONLY way to fix it is a ACC cal and that only fixes it for ~2min.

CerealKiller1, 2016-04-15:
Ok well set acc_cut_hz=4 didn't do anything, actually it was already at 4 so I tried messing with it and no change visible changes to flight behavior. 

For now I've reverted to a known working 1.11 and matching backup / dump. Will keep following this issue as I'd prefer to use 1.12 cause of the failsafe settings.

lit1088, 2016-04-15:
what is your test results ？set acc_cut_hz=4，Whether it works well and   no drift   issue in ANGLE and HORIZON ?

CerealKiller1, 2016-04-15:
Results were no change. It was actually at 4 already, I tried values (every 2) from. 0 to 14 with it not making any visible changes. 

I am out here flying now and have my full setup includig computer and am happy to test more stuff... I am subscribed and should get an email notification if you respond again and would like me to try somethig else. 

Again sorry, I just in how hopeless it is without Blackbox :(

DeetAus, 2016-04-16:
I just wish I could roll back my firmware to 1.11, but in this case I have an SPracingF3Mini and 1.12 is the only version available

This really sucks, I was really happy with my Cleanflight, really happy with my original SPracingF3 board, but this now has been a serious letdown

[43 more replies]

closed 2016-05-07 by hydra