cleanflight/cleanflight #1793: Buzzer doesn't work on SPRacing F3 deluxe.

drichards0725, 2016-02-08:
Seems there is a difference on SPRacing F3 deluxe and the use of the buzzer on the buzzer pins not working.

pdoreika, 2016-02-10:
I have found that a fairly large amount of counterfeit SPRacingF3's are being made and sold by Chinese manufacturers.  In fact, I was told that if you bought an SPRacingF3 from a Chinese seller then you are practically assured to have a counterfeit.  I learned this AFTER I bought one over the internet.  The good news is that the buzzer will work, but you'll have to solder to the pins on the bottom of the board.  I got mine to work fine that way. The other good news is that it appears to otherwise work just fine.

drichards0725, 2016-02-10:
This was purchased from getfpv and does have the cleanflight logo on it.

pdoreika, 2016-02-10:
It's probably NOT a counterfeit board then.  Though I've never heard of a buzzer going bad I guess it's possible.  Have you checked polarity?  Tried a different buzzer?  I do not know if there is a way to suppress the buzzer in the Cleanflight software.  I presume you have put 5 volts to the board?  If you're using ESC's without a UBEC (like the OPTO ones I'm using) then a separate power source has to be applied to the board.

If it turns out to be a defective board I'm confident that getfpv will work with you to replace it.  I haven't bought from them before but have heard that they are very reputable.

drichards0725, 2016-02-10:
Yes I have ordered and tried other 5v buzzers that worked with 5v applied
and I am running the lumenier 20amp esc with oneshot enabled connected on
the pdb from getfpv. I normally run the cppm mode and tried the pwm
thinking it was an issue but still didn't change the buzzer working. The
only difference in my thought was this was the deluxe spracingF3 and
whether it made any difference.

pdoreika, 2016-02-10:
I wouldn't think that the deluxe version makes any difference but unfortunately I am still pretty new to the hobby and I am at the limit of my ideas on the matter.  Hopefully someone else will chime in but in the meantime, if I were you, I would drop a note to getfpv about the issue.  They may have the answer, or be willing to replace the board for you.  Either way it can't hurt.  Sorry my friend.

drichards0725, 2016-02-10:
Thank you for the help and yes I already reached out to them and they
didn't have any answers either on it. I am waiting for my new spracingf3
mini to arrive to swap out and see if they will replace this one as I need
to practice for racing.

Thanks and have a wonderful day. :)

drichards0725, 2016-02-13:
Just checked with Naze32 rev5 deluxe and Buzzer does not work as well and I have 5 brand new buzzers that i checked with that are 5v piezo buzzers from hobbyking that work on the kk2 board but on on the most recent firmware. Something in the firmware leads me to believe that the buzzer header pins on both the SPRacing F3 and the Naze32 rev5 don't work as a result of a code change. I am waiting for another SPRacing mini FC to arrive so i can swap out and test to eliminate deluxe vs non-deluxe FC's.

drichards0725, 2016-02-15:
Issue was resolved due to jst connector pins being slightly bigger by .01 microns  then the connector header pins on the board. Compressed jst connector pins slightly to make contact and works. Please close this issue.

closed 2016-02-15 by drichards0725