cleanflight/cleanflight #1794: 1.12.0 RC2 default for Naze32

3djc, 2016-02-09:
Pre RC2, default for Naze32 with Oneshot ESC where resulting in a 500Hz pulse frequency for the ESC. Applying RC2 default result in gyro_sync (not an issue per say) but also a pulse frequency of 1 khz, which many ESC do not handle well. Changing gyro_sync_denom to 2 bring it back to  500Hz, closer to previous defaults, fixing it for me (at least on the bench). Will fly test it when the tempest is over ....

borisbstyle, 2016-02-09:
What ESC doesnt like 1khz? As far as I know there is no ESC out there what has problems with it.

3djc, 2016-02-09:
Hello Boris,
I use ZTW Spider Series 18A, blheli 14.4. When running rc2 défaut, i have a stable 1khz out pulse to the esc, but the esc will reset within 10 or 20 secs. If i turn off i2c overclock, i have a more jittery output, between 977 and 946hz. The esc will reset but it takes longer. If i set the gyro divider to 2, i get a 500hz output, and the esc does not reset. I do not have a signal generator at hand to find the exact cuttoff tho :(

borisbstyle, 2016-02-09:
Me and many others have tested pretty much any ESC on the market and never experienced that. 

Are you sure that 1khz is the reason?

3djc, 2016-02-10:
Hello Boris. You are right, I have done further test and the 1 khz is not the issue, but a side consequence, my apologies for getting us on the wrong track.

On my FC, the issue is that the default settings seem  to work fine ONLY when nothing else is activated. If you enable GPS it fails, if you enable softserial, it fails,(that's just on the bench, I've not taken it flying !)

If you set the divider to 2, you can enable those without failing it

closed 2016-02-23 by 3djc