cleanflight/cleanflight #1795: I2C overclocking seems to be set on by default

martinbudden, 2016-02-09:
See:

https://github.com/cleanflight/cleanflight/blob/master/src/main/config/config.c#L492

This may be the desired behaviour, I raised this issue in case it is an oversight.

Also note looptime is set to 2000 by default. Just checking this is correct.

3djc, 2016-02-09:
About the looptime, it is not used I believe, since gyro_sync is enabled by default

borisbstyle, 2016-02-09:
@martinbudden 
Yes correct. Higher i2c speed does not have any negative effects on the boards out there. Why just not enable it. This allows more optimal defaults for F1 boards

martinbudden, 2016-02-10:
OK, I'll close the issue. I saw this and just wanted to check it wasn't done accidentally.

borisbstyle, 2016-02-10:
It is totally not a problem to have high speed i2c running on small
electronic circuits with just 1 or few components connected.
The trouble would only start with many devices connected or longer
distances between them.
High speed i2c still stays within the voltage specs so the chip can
recognise rising/falling edge.

ledvinap, 2016-02-10:
@borisbstyle:
It's not that easy - I2C uses filters to minimize noise influence. Standard defines 100/400kHz and filters may be optimized at this frequency. Some chips may not work at higher speed at all. But even worse, some filters may be set to about double of 400kHz, so running at 1MHz may be at edge, working only for someone ...

Also MPU is not the only peripheral on I2C bus, other chips may have problems ..

borisbstyle, 2016-02-10:
@ledvinap 
I understand that.

I did extensive testing on a full list of available Flight controllers out there right now. 
I found the worst performing F1 flight controller to be naze32 rev6.....not sure why. Maybe the baro type. That one had limit of 1Mhz. All other targets were able to run up to 1.2Mhz without issues.

I decided to put a safe limit to 800Khz on F1 targets and 1Mhz on F3 targets. Till now i didn't find any issues in my "quite large" beta test group.

ledvinap, 2016-02-10:
Yes. Just commenting on distance part - there are other things that limit speed.

borisbstyle, 2016-02-10:
@ledvinap 
Yes I know :d. I am trying to prevent bad stories going around about it :d

It does seriously unlock good performance on some boards like naze32. 1khz full gyro sampling for example does make a difference.

martinbudden, 2016-02-10:
Reopened because there is an ongoing discussion.

martinbudden, 2016-02-10:
My view:
For people who want to get the absolute max performance out of their boards, setting overclocking on is just one small extra thing to do.

For people who have just built a copter and want to get it going, this is something that could go wrong. Admittedly it probably won't go wrong, but if it does it is the kind of thing that could take ages to figure out.

Remember we are talking about the default here. In my view the defaults should be the "most compatible" option, not the "highest performance" option.

closed 2017-02-25 by hydra