cleanflight/cleanflight #1807: Oneshot125 doesn't work with naze32 and BLHeli

copter58, 2016-02-10:
In Cleanflight 1.12.0 RC2 Oneshot125 does'nt work with naze32 and ESC "Favourite Sky 3 Quattro 4 x 12A  BLHeli"
[http://www.hobbyking.com/hobbyking/store/__69363__Favourite_Sky_3_Quattro_4_x_12A_Brushless_Quadcopter_ESC_2_4S_1A_BEC_BLHeli_EU_Warehouse_.html](url)
In the Motor-Tab i can see the Motor-Signal, but the Motors doesn't turn. 
With 1.12..0 RC1 it work's.

borisbstyle, 2016-02-10:
@copter58 
There is no issue with the RC2. You just have outdated ESC firmware.

you can still use it though by decreasing looptime:
set gyro_sync_denom = 2

Note that its better to upgrade your ESC's to firmware what does support modern looptimes.

copter58, 2016-02-11:
Thank you. With BLHeli 14.4 it works well

SteveAmor, 2016-02-15:
@copter58 can you please close this issue.

closed 2016-02-15 by copter58