cleanflight/cleanflight #1811: RC Inputs 5-8 outputs PWM under RX_SERIAL or RX_PPM

jflyper, 2016-02-10:
Cleanflight 1.11.0 and 1.12.0, target is NAZE.

The RC inputs 5-8 generates (outputs) constant PWM (1ms high, 1.5ms low) after reboot, when receiver mode is set to RX_SERIAL or RX_PPM.

I initially found this with no features enabled.
If the SOFTSERIAL is enabled, the behavior is completely suppressed (naturally).
If the LED_STRIP is enabled, the behavior is completely suppressed, although LED_STRIP only uses RC5.
If the SONAR is enabled, the PWM output stops on RC7 and RC8 after 2 seconds, while RC5 and RC6 continues to generate the PWM output.

Are these all expected behaviors?

sigrokBlack, 2016-02-11:
Could be linked with my issue: https://github.com/cleanflight/cleanflight/issues/1813

jflyper, 2016-02-20:
I did some tests with previous releases, and found out the phenomenon already existed in the RELEASE 1.0.0 NAZE 2014-11-25 15:53 (stable).

1. Flash 1.0.0 NAZE 2014-11-25 15:53 (stable).
2. Connect with the 1.2.0 CFC.
3. feature -RX_PARALLEL_PWM
4. feature RX_PPM
5. save
RC Inputs 5 through 8 outputs PWM.

Although the phenomenon is not fatal for me atm, I just feel uncomfortable something called inputs generating outputs.

jflyper, 2016-02-22:
I did some further investigation, and found unused RC inputs and PWM outputs are actually initialized as motor outputs in pwmInit() in drivers/pwm_mapping.c.
It still makes me uncomfortable because I usually set unused I/O pins to inputs if pull-up/down are available (don't know if stm32 has configurable internal pull-up/downs), but I will close the issue for following reasons.
(1) It will require substantial work and knowledge beyond my current capacity to modify pwmInit() to detect and configure actual motor configuration and initialize only the corresponding pins.
(2) People don't seem to have much problem with the current code/behavior.

Oh, BTW, in the course of my investigation, I found a bug in init() that prevents SONAR pins from being detected by the pwmInit(), which caused the initial pwm signals on sonar pins. I will open a new issue for this one.

closed 2016-02-22 by jflyper