cleanflight/cleanflight #1813: First 4 Input Channel not working, tested with multiple boards

sigrokBlack, 2016-02-11:
HI,
I've two Naze32 boards, 6 DOF and 10 Dof, since Cleanflight update 1.10 I'm not able to use Inpu 1, 2,3 and 4 anymore. My Aux channles are working, is there a bug with the mapping of the channels?

Arakon, 2016-02-12:
Actually, that behavior points to failsafe. It sounds like your stick channels are in the wrong order, missing, or out of bounds (1000-2000).

sigrokBlack, 2016-02-12:
No, i don't think so. My Failsafe is turned off and the Aux channles are working great. When I use an Arduino to simulate an PPM signal and chabge the cleanflight settings to ppm its working great on channle 1 aka PPM input. So the Gpio are not broken.  I'll hock up an logic analyzer to rx pins 1 to 4 when in rx_pwm_parellel to see if the ports are used for something else.
How can I change the pinout? How I said v1.9 is working 1.10 not. Some pinouts must have changed or?

joshuabardwell, 2016-02-12:
I believe rx_min_usec and rx_max_usec changed between 1.9 and 1.10. Check that your channel endpoints are not going outside those values.

sigrokBlack, 2016-02-12:
Error was an faulty solder joint between the header pins of my Naze32.
Sorry, I thought it was an issue because it appeard on two Naze at the same time(the secound naze has a broken solder trace) Everything is working. Thanks for your support!

closed 2016-02-12 by sigrokBlack