cleanflight/cleanflight #1814: RX_PPM - Receiver not working with oneshot125 enabled SPRacingF3 Mini

Lippy1968, 2016-02-11:
have tried 

Flashing the receiver with 27ms Firmware.
Re-flashing the controller with the same firmware using DFU
Setting failsafe on TX. 
Resetting RX
If oneshot is disabled, then the controller sees the RX input and enabling FC with TX gives full functionality

borisbstyle, 2016-02-13:
It all works here.

Are you sure this isnt your esc not able to do fast oneshot?

Lippy1968, 2016-02-13:
on the frame are Afro 20a Race spec opto ESC's. I have flashed them with BLheli and Simonk and neither firmware works, both exhibit the same issue. I can try moving the FC to my other quad with Littlebees in and see what happens.

borisbstyle, 2016-02-13:
So you dont see your channels moving in RC tab in the configurator?

Lippy1968, 2016-02-13:
No channels move, remains static (with the TX on or off I usually have jitter).I have flashed a spare Naze rev6 with 1.12RC6 and put that in the same frame with the same hardware used on the F3 mini. Enabled oneshot and that works as expected.

3djc, 2016-02-15:
I have just received my SP F3 mini this morning, and I have the same issue.

I have connected only the 4 esc and ppm input, absolutely nothing is configure with the board.

Set receiver to PPM, oneshot disabled, RX pane show normal response to RC input. Oscilloscope test on ESC input show regular PWM

Change nothing, just set oneshot to enabled, reboot : RX pane is not showiinng the sticks movement anymore,  Oscilloscope test on ESC input show onshot pulse

3djc, 2016-02-15:
I have restested a little differently for ease of reproduction, but face the same issue :

SP F3 mini connected just to USB and receiver
Flash  1.12 RC2
Connect CFgui, by default RX is PPM, oneshot diabled, go to RX tab, things are moving
Go to config pane, enable oneshot, save and reboot
Go to Rx pane, no more movement

Lippy1968, 2016-02-15:
Are you using Opto esc's?
I have tried numerous options. Different esc's. Minimal configuration yet everytime is exhibits this issue. I am wondering if the issue reported as a bug in the sparky and cc3d is similar.

3djc, 2016-02-15:
As said in my previous post, you don't even need the ESC  connected to exhibit the issue, ESC have nothing to do with it

borisbstyle, 2016-02-18:
Guys sorry for my confusion above. I never tested the mini and PPM. I initially thought it was about the normal SPF3

Lippy1968, 2016-02-18:
No problem. Initially I thought I had broken mine. I looked at some of the timers, but was unsure as to which would clash. I did some changes and tested but never got any further.

Lippy1968, 2016-02-19:
Tested working on my board. Thank you all for sorting this. Cant wait to go for a test flight with this board!

hydra, 2016-02-19:
Fixed.

Thanks for bug reports everyone.

Build 3096 has the fix and cleanup applied.

https://travis-ci.org/cleanflight/cleanflight/builds/110385093

Binaries for testing here:

http://cleanflight.memoryleaks.org/builds/cleanflight_cleanflight/2016-02-19/3096/

closed 2017-02-25 by hydra