cleanflight/cleanflight #1817: voltage sensor fluctuation

Cheredanine, 2016-02-12:
Lux target, CF1.12 RC2 (4s)
(confirmed on BF2.4 RC7 on naze and lux also)
VBat Voltage reading fluctuates rapidly with about 1 volt between peak and troughs
Initially noted in flight as low voltage alarm starts and stops early
testing on the bench just adding battery the symptom can be seen on voltage reading of the setup tab even immediately after clean flash

CF1.11 does not have the issue
BF 2.4 rc8 has attempted to fix and has reduced the magnitude from 1V to about 0.5V but still fluctuating
(aware DE has made a comment on the code)

borisbstyle, 2016-02-13:
yes this most likely has to do with the new scheduler causing a lot of jitter for battery readings.
The filters are sensitive for jitter.

i have a pr for new filtering of battery voltage....i think a median filter is necessary too on top of the existing one.

borisbstyle, 2016-02-13:
@hydra 
I will try to fix this in PR https://github.com/cleanflight/cleanflight/pull/1806

ledvinap, 2016-02-13:
@borisbstyle : median filter is nonlinear - it will cause more problems than it can fix.

ADC is read using DMA, it's quite easy to change the initialization so that queue of ADC values is available. ADC task may then filter them (or just use mean filter and implement summing in ADC driver)

borisbstyle, 2016-02-13:
@ledvinap 
I am not really familliar with ADC driver part. Can you link the particular code?

borisbstyle, 2016-02-15:
I tested the vbat yesterday on biquad pr from #1806 and it just works stable as before

hydra, 2016-02-18:
@Cheredanine please test latest build and report back.

hydra, 2016-02-18:
I just tested latest code on an SPRacingF3Mini and VBAT monitoring seems ok there.

Cheredanine, 2016-02-18:
@hydra sorted mate, dropped to 16.1 did move up to 16.2 over about 10 seconds but that could be voltage recovering after boot of escs etc, I would suggest you can lose this

Cheredanine, 2016-02-20:
Closing, seems to be fine in latest

closed 2016-02-20 by Cheredanine

vzachari, 2016-09-29:
Apologies if this is the wrong place for this. I am getting VBAT fluctuations within +-0.5V with the Lux target, very similar to what initially reported for #1817. This is on CF 1.13.0. The fluctuations are unaffected by clock speed.