cleanflight/cleanflight #1818: gyroADC has a dangerous overflow scenario (int16_t )

borisbstyle, 2016-02-13:
Just copying discussion between me and @ledvinap from other not related PR.

I discovered that gyroADC has no limit. That means that when quadcopter configured in certain way with very high rates can cause inverted gyro data. I managed to get that in the log as well. So above 2000deg/sec gyro goes completaly opposite what can result in infinite rolls. Where do you think it would be best way to protect that.I would say each PID controller should do some kind of limiting?
@ledvinap Cleanflight member
ledvinap added a note an hour ago

(new issue may be adequate)
So there is gyro overflow? .. Where?
Gyro chip should return 13bit signed value (-4096-4095) with saturation and overflow indication.
Rest of code may use saturation logic arithmetic to handle overflow ...

Also there may be problem with gyro saturation - it may cause PID i-term growing when command is larger that gyro measurement range. Constraining commanded input within gyro range can help
@borisbstyle
Cleanflight member
borisbstyle added a note an hour ago

Check the end of this log where infinite roll occurs. Note that this user was using unusual high rates. negative gyro value rolls into positive there beyond 2000deg/sec.
LOG00436.TXT
@borisbstyle
Cleanflight member
borisbstyle added a note an hour ago

Yes it's a good idea to constrain the error as thats where all things come together. rcCommand /rates etc
@ledvinap Cleanflight member
ledvinap added a note 9 minutes ago

Was wrong - output range is 16 bit.

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/gyro.c#L128

gyro offset must be applied with saturation - there is overflow with 16bit values.

Also using biquad filter may result in overflow (https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/gyro.c#L145)

IMO it would be best to either use int32 for gyroADC or to scale gyro reading down (divide by 2 or 4) so that there is some safety margin.

ledvinap, 2016-02-13:
@hydra: Quick fix is using saturation arithmetic, but it will not avoid new bugs in future.

What about: 
- int32_t gyroADC
- divide gyro reading by 2 to get safety margin

ledvinap, 2016-02-13:
BTW: in linked log file, ADC Y and then Z gets saturated too (and overflows to opposite sign too). Is that really caused by centripetal force? The data looks like almost 3000RPM body rotation ... 

[image]

ledvinap, 2016-02-13:
(and ACC reading is also affected by overflow)

borisbstyle, 2016-02-13:
yeah this was crazy high rate. Not something average users will use, but it
reveals the issue.

borisbstyle, 2016-02-13:
int32_t sound like the best thing. divide by 2 would reduce resolution, but
wouldnt be that much of an issue.

hydra, 2016-02-13:
interesting..  I had a case just today where my quad started doing infinite rolls into the ground (snow, so no damage) as you noted. very timely!

was quite spectacular.  I was flying CF1.12.0-RC2.

hydra, 2016-02-13:
my current vote, is int32_t.

borisbstyle, 2016-02-13:
Just configuring  your rate numbers to do about 6 rolls per second can
reproduce it.

It also can happen when one of the ESC's or motors would fail. Not that you
can safe it, but you may make better chance with correct gyro data.

Its obviously something needs to be fixed :). int32 is def the most
suitable solution....it fits rhe cpu architecture the best anyway

bdoiron74, 2016-02-13:
Also, any commanded rate fed into the pids should be limited to avoid
exceeding the actual sensor max, if not already.

kgustine, 2016-02-13:
I was reading over the airmode implementation (#1766) and came upon your discussion on this issue.  I believe I have come across this many times and never knew why the quad took so long to come out of a roll.  

My use case is that I use throttle compensation in Angle mode (on my Tx).  This is kind of cheating because I don't have to worry about altitude when giving excessive pitch or roll.  I actually apply 60% of pitch + 60% of roll to my throttle. Max angle is set to 60deg.  It works amazingly well.  Anyway, the fun part is that you can go to min throttle up in the air and give a quick kick to the pitch or roll and, due to the throttle compensation, it will cause a really fast (super fast) rotation in free fall.

Sometimes it doesn't come out of it so gracefully, but I do this 50+ meters in the air.  Sometime it takes a full two seconds to stable out once I give it throttle.  Does it sounds like a symptom of the gyro overflow issue?

SteveAmor, 2016-02-13:
This is interesting to me given the fix I implemented in the original KK2.1 code where the coders thought they set the MPU6050 to 500 deg/sec when in fact their code didn't work and it stayed at the default angular rate of 250 deg/sec. 

To me, the answers to this issue is to stop the pilot commanding an angular rate of more than 2000 deg/sec.  I'm not convinced there is an "catch all" answer to gyroADC hitting 2000 deg/sec with CF supporting so many gyros.

ledvinap, 2016-02-13:
There are two problems:
- gyro saturation: It is reasonable to allow command larger that max gyro rate(quad is capable of higher rate, user should know what is he doing), but limit its influence only on P and D terms, preventing I term grow and resulting overshot.

- `int16_t` overflow when gyro offset is applied. This happens only in one direction (based on gyro offset sign) and results in change of sign of rate variable. Perceived rate changes from maximum positive to maximum negative. Command direction do not change, so maximum control output is applied (keeping gyro in saturation), even when sticks are centered again.
Using `int32_t` will fix this.

bdoiron74, 2016-02-13:
The problem with letting the gyro saturate is that the imu will simply fall
behind the actual orientation. It'll then really on the acc to correct it,
which as Kenneth noticed, takes time due to filtering and also requiring a
stable acc vector.

Not a problem for those flying pure rate mode only, but a big problem for
the guy flying horizon mode or trying to make a save by flipping back to
level mode.

I'd saturate the commanded rate to something less that the sensor maximum
to leave a little headroom.

borisbstyle, 2016-02-13:
@bdoiron74
Both approaches do make sense. Perhaps level modes should even get lower
limits as acc is much slowlier.

Anyway.....who is going to do a PR?

ledvinap, 2016-02-13:
@bdoiron74 : Good point on gyro integration.

But someone who sets stick rate high enough to get 2000dps should know what is he doing ;-)

In long term, approach like Kalman filter may be used to keep track of  how accurate is position estimate. IMU may than set position estimate reliability to zero on gyro saturation. ACC input may be used to estimate orientation (but that will be probably useless at this rotation speed). IMU convergence can be very fast when gyro data are available, possibly fast enough to save quad ..
Any volunteers?

borisbstyle, 2016-02-13:
I would first start with a quick fix to do int32 only before the 1.12
releases.
Thats an easy part and should work I guess.

Later the other stuff can be done.

Without disrespect....the guys eho use horizon/level will never be using
rates as high as this.

bdoiron74, 2016-02-13:
Does the configurator tell you what your full stick rate is? Maybe popping
up a warning would be good for those who might do it by mistake. I guess it
would then have to know what gyro is in use if cleanflight supports gyros
with different ranges.

ledvinap, 2016-02-13:
```
/accgyro_spi_mpu6000.c:276:    gyro->scale = 1.0f / 16.4f;
/accgyro_l3g4200d.c:74:    gyro->scale = 1.0f / 14.2857f;
/accgyro_spi_mpu6500.c:139:    gyro->scale = 1.0f / 16.4f;
/accgyro_l3gd20.c:160:    gyro->scale = L3GD20_GYRO_SCALE_FACTOR;
/accgyro_mpu6050.c:80:    gyro->scale = 1.0f / 16.4f;
/accgyro_mpu3050.c:64:    gyro->scale = 1.0f / 16.4f;
/accgyro_mpu6500.c:62:    gyro->scale = 1.0f / 16.4f;
```
seems like ~2000dps is used in current gyros ...

MasterZap, 2016-02-14:
> Not a problem for those flying pure rate mode only, but a big problem for
> the guy flying horizon mode or trying to make a save by flipping back to
> level mode.

I tried exactly that, coz apparently, 75% acro plus is... a tad much.  I have a muscle memory flipping into a stabilized mode to get out of whacky  binds when I can't tell what's up, so muscle memory kicked in and did exactly that. However after the crazy flipperoo, quad thought down was up....

https://www.youtube.com/channel/UCouubroTfp_0izKFjaWZ0KQ

So yeah, there are people that do fly high rates, but do have a level mode in there for "safety". Actually I'd say thats the case you really need one :)

/Z

MasterZap, 2016-02-14:
> But someone who sets stick rate high enough to get 2000dps should know what is he doing ;-)

Well, there is the wee problem that CleanFlight/BetaFlight has no sensible units (or if it has, it's undocumented). Nobody actually knows what the various rates even mean. At least there is where OpenPilot was helpful where you actually set this stuff in degrees/second....

And of course, Acro+ makes it impossible to know what's gonna happen... just sayin'... (see above video :) )

/Z

bdoiron74, 2016-02-14:
My thoughts exactly - I remember sifting through many layers of code just
to figure out what my settings might mean in the real world.

borisbstyle, 2016-02-14:
Just played with int32 and it fixes the issue. 
It's quite easy to reproduce by using configurator as debug tool. Divide gyroADC by 2 to make sure that debug parameter doesnt overflow and try both scenarios. Just twist the board very hard to reach 2000deg/sec. Quicky twisting it from left to right by holding USB header can reproduce it easily.

Current case. You can clearly see the overflow occuring.
[image]

WIth int32 fix.....no more inversions
[image]

The code I use is pretty simple. I left the driver part untouched and changed that one to gyroADCRaw (int16_t). gyroADC is the converted int32_t value and that one is advertised globally. 
I also did the same for acc and mag as they all share same board alignment code.

int16_t gyroADCRaw[XYZ_AXIS_COUNT];
int32_t gyroADC[XYZ_AXIS_COUNT];

gyroupdate part
    // range: +/- 8192; +/- 2000 deg/sec
    if (!gyro.read(gyroADCRaw)) {
        return;
    }

    for (axis = 0; axis < XYZ_AXIS_COUNT; axis++) gyroADC[axis] = gyroADCRaw[axis];

    all post processing with gyroADC happens after this part 

@ledvinap What do you think of the above fix?

borisbstyle, 2016-02-15:
I have a fix here. Please review before I make a pr. Unittests deem broken in the master so that fails anyway but I think the unittests will work too.

https://github.com/borisbstyle/betaflight/commit/2340200fa017ae119c4cf6b238b1f345ce53992d

borisbstyle, 2016-02-15:
and some real flight testing by the same guy who provided the blackbox data on same rates and even higher. Couldnt reproduce it anymore.
https://youtu.be/t1ZmZ9i4jjc

hydra, 2016-02-16:
fix merged.

closed 2016-02-16 by hydra