cleanflight/cleanflight #1821: spracingf3 mini bootloader not functioning

CFBronson, 2016-02-14:
I successfully jumped boot pins and entered bootloader however; i cannot flash spracingf3 mini firmware to board. Followed procedure given in CF, get message ;Failed to open serial port.  .

3djc, 2016-02-15:
No issue here flashing mine with 1.12 RC2 mini firmware. Are you sure you followed the manual procedure (STM drivers then Win USB DFU drivers ?). Mine was ok to flash naze32, but not the mini until I followed the procedure.

CFBronson, 2016-02-15:
yes i finally read that little gem buried in there. I didnt realize i had to flash from dfuse. I just flashed 2.4 today and so far so good.

closed 2016-02-15 by CFBronson

teddraketitusville, 2016-02-23:
I have the SPRacing Mini and cannot get it to connect.  I have two boards.  One connects, the other doesn't.  I'm trying to do recovery.  Any pointers.

CFBronson, 2016-02-23:
Did you follow the instructions for flashing a DFU file? It's not going to
flash through cleanflight.

teddraketitusville, 2016-02-23:
One Mini connects and I have upgraded to the latest firmware right by the instructions in your user manual.

The other won't connect.  Activity light does flash.  What do you suggest?  Short boot pins and reload?  Does the button work for boot?  Any additional information would be helpful on how to reload.

It's just odd that one worked perfect and the other doesn't.

Thanks.

Ted Drake

3djc, 2016-02-23:
The mini DOES flash from the cleanflight gui, you need to follow the install instructions, which are not straightforward to find !!!

The manual tells you to install the STM32 virtual com port. Most people just run the setup and do NOT read the readmi inside the zip. That readme says : 

1- Uninstall previous versions (Start-> Settings-> Control Panel-> Add or remove programs)

       2- Run your "VCP_V1.4.0_Setup.exe" 
	   
	   3- Go to Your installation directory - Example, C:\Program Files (x86)\STMicroelectronics\Software\Virtual comport driver
	   
	   4- Go to Your OS version directory ([Win7] or [Win8])
			+ Then :
				- Double click on dpinst_x86.exe if you are running a 32-bits OS version
				- Double click on dpinst_amd64.exe if you are running a 64-bits OS version
			+ Follow the instructions.

Step 4 is often forgotten, and that' s why some cannot flash using cleanflight client.

Ted, yes your issue is strange. When you say won't connect, is it a flashing issue, or no connection using USB at all ?

teddraketitusville, 2016-02-23:
Won't connect at all.  Gives all the lights like it's fine.  Click connect and it won't connect.  Shows a fail after 10 seconds.

I bought two boards at the same time.  Same SPRacing Mini F3.  One works just like the manual says.

Ted Drake

CFBronson, 2016-02-24:
Ok , it not my manual I didn't write it. I may be able to help. The buttons
can be used for boot mode.  Hold both buttons down and then power quad or
plug into your pc or whatever you use to load cf(or your battery) ,
continue holding both buttons for about 5 seconds and release. Red led
should go off when you release or you didnt do it right. If the red doesn't
go out make sure you are holding both buttons and count slower lol. I've
also got some feeling that this is the same process foe binding the sat rx
to the board. It would be the same process but with only holding the B
button instead. I would test it for you but I unfortunately reversed input
polarity on my spracing f3 mini today whilst soldering in a new PDB
-Instant white smoke. I guess I was to tired to be playing around.
The only other thought I have is if you are using correct firmware
adapter.  I could only get mine to work by converting my .hex file of
betaflight 2.4 to a .dfu file. There's a tutorial in the manual under
flashing.  You can YouTube the video and download the dfuse program and
follow that path. Anything else let me know . Good luck happy flying!
-Charles.

teddraketitusville, 2016-02-24:
I have downloaded DFU Manager, created a DFU from the HEX and used DFUSE to
load it directly to the chip.  Then recycled and gone back to CF and it
will not connect to CF.  The one that I have that works fine comes up with
COM1, COM7 and Manual Selection.  I choose COM7 and it connects.  The bad
one only shows COM1 and Manual Selection and will not connect on COM1.  Any
more ideas?

CFBronson, 2016-02-24:
Send it back?  I'm sorry but I think out might be bad.  You used the button
boot method successfully right?  I thought that was super helpful. Thoes
boot pins are tiny.