cleanflight/cleanflight #1822: spracingf3 mini buttons not functioning manual needs update

CFBronson, 2016-02-14:
The manual does not describe the usage procedures for on-board buttons "A" and "B"

CFBronson, 2016-02-15:
Further work with this board today. I found that boot mode can be entered by following this procedure: 1.with craft unplugged from computer and not powered by battery press and holding both buttons while plugging in power from either sources.  2. Hold for ~5 seconds and release.  This will give you boot mode.

davidh-, 2016-03-20:
@CFBronson you are right, it indeed does work. Try to hold for at least 5 seconds minimum. Holding for anytime between 5-10 seconds will work actually:

```
// buttons released between 5 and 10 seconds
    if (secondsRemaining < 5) {

        usbGenerateDisconnectPulse();

        flashLedsAndBeep();

        systemResetToBootloader();
    }
```

src - 
[https://github.com/cleanflight/cleanflight/blob/c00fd2ab9baff347e4e746df3f1b1ae250f65a6e/src/main/main.c#L215](url)

@hydra please mark this as resolved (I myself do not see an option to mark it as resolved). Thanks!

closed 2017-02-25 by hydra