cleanflight/cleanflight #1842: Allow the direct use of potentiometers for in-flight PID tuning

grokasm, 2016-02-17:
The current in-flight adjustments functionality is greatly configurable and powerful and allows incremental modification of a range of values from PIDs to rates in-flight using 3-Position switches.
There has been a lot of interest in the brushed micro community to improve the current in-flight adjustment implementation to allow for the direct setting of these values (as opposed to delta adjustments). 
For example, use a pot on an AUX channel to directly set the P values for pitch and roll allowing you to literally "dial in" the proper values.
This in flight adjustment would enable the pilot to more easily get an intuitive feel for how the various values affect flight characteristics and stability.
A greater explanation of this method of tuning can be found [in this thread from the micro motor community](http://community.micro-motor-warehouse.com/t/alienwii-in-flight-adjustable-pids-tutorial/219).
I have already started looking at the code base, in particular rc_controls.c to attempt implementation.

Also, I was unsure what the best approach would be for adding this functionality? I'm divided between two possibilities:
1. Add a parameter for each adjustment range that switches between incremental/absolute. That way each adjustment function remains as is and a new switch/dropdown would need to be added to the configurator.
2. Add new adjustment functions for each value that works differently. For example, there would be `ADJUSTMENT_PITCH_P` for the current incremental version and `ADJUSTMENT_PITCH_P_ABSOLUTE`/`ADJUSTMENT_SET_PITCH_P` for the direct setting version.

It would be great to get some feedback from the community about this particular feature.
Would this be useful to you? Is there any reason why this would be a horrible idea?

complacent1, 2016-02-17:
I am very interested in this. I have come to depend on adjustable pids with multiwii by using the side sliders on my Taranis. You just can't do that with the current implementation of adjustments in Cleanflight. I would also like to see adjustments for LEVEL be available for those of us that fly Horizon mode and want to adjust those pids with a potentiometer as well.

ericfairbanks, 2016-02-17:
This would be excellent, and the absence of this feature is the #1 reason I hear from people who prefer 328p/multiwii FCs over baseflight/cleanflight.

The ability to switch between incremental and absolute would be really handy, as setting PID parameters with only relative adjustments could easily spiral out of hand with no way to reset to an absolute value.

Also +1 for horizon mode adjustments.

pilsnerpopper, 2016-02-17:
👍
Tuning PIDs is my least favorite part of the hobby.  Being able to do in in-flight would be a huge improvement.

joshuabardwell, 2016-02-17:
You can also do it incrementally in-flight.

If you intend to do Absolute values off a pot, IMO you must give configurable endpoints. Otherwise, for example, you will be tuning P from 0.0 to 25.0, which is clearly incorrect, and would not give the required resolution in the range of interest. Likewise, you would be tuning I from 0 to 255--same situation.

joshuabardwell, 2016-02-17:
Possibly, it would be acceptable to have the entire channel range map to the entire parameter range, and then use the mixer configuration in your transmitter to set the endpoints. This might actually make more sense than building the endpoints into Cleanflight, but it would be harder for users to implement, and some users would not be able to implement it if their transmitter didn't support it.

gpolic, 2016-02-17:
That is a nice idea.
The Openpilot/Librepilot have this and they have configurable Min & Max in the aux channel
You can adjust your pot in the middle position before takeoff, and take it from there
Obviously the user must have some idea of the values range

lostcontrol, 2016-02-17:
+1 that's definitely a feature I miss in Cleanflight. APM:Copter has it and it is IMHO better that relative adjustments. There also you have to define min/max in the FC.

JG101, 2016-02-18:
This would be a step forward. I cant't offer much other than a willingness to test I'm afraid.
I am with complacent1 in thinking that the ability to adjust the LEVEL strength would be a boon for FPV fliers and people weaning of stabilised modes.

lostcontrol, 2016-02-18:
What if the potentiometer is used to adjust the value around the currently saved value? Not sure I'm clear so let's take an example:

Your saved RC rate is 1.00. The effect of the pot is hardcoded (or make it configurable) to e.g. 20%. Moving the pot from min to max would allow you to change the setting from 0.80 to 1.20. If you are a bad pilot like me and notice that 0.80 is still too much, you land with the pot at min, save with stick combination and move the pot again in the middle. You can then adjust from 0.64 to 0.96. Etc, etc, etc...

I see here the advantage of not adding additional configuration parameters for min/max and the ability to work with small or big values. A critical issue I see is e.g. saving value, forgetting to put the pot back to middle, saving, forgetting to put the pot back to middle, saving, etc. Also it would require to save the currently saved value somewhere since I guess accessing the flash during flight is not a good idea.

jarno83, 2016-02-18:
I like that idea. Lostcontrol method would be great 😀 hope it's not hard to code. Thanks.

grokasm, 2016-02-18:
@complacent1 @JG101 The level strength and rc rates and any of the other adjustment functions currently implemented can have this absolute setting functionality.
Here is the list of adjustments that can be set up:
```
    ADJUSTMENT_NONE = 0,
    ADJUSTMENT_RC_RATE,
    ADJUSTMENT_RC_EXPO,
    ADJUSTMENT_THROTTLE_EXPO,
    ADJUSTMENT_PITCH_ROLL_RATE,
    ADJUSTMENT_YAW_RATE,
    ADJUSTMENT_PITCH_ROLL_P,
    ADJUSTMENT_PITCH_ROLL_I,
    ADJUSTMENT_PITCH_ROLL_D,
    ADJUSTMENT_YAW_P,
    ADJUSTMENT_YAW_I,
    ADJUSTMENT_YAW_D,
    ADJUSTMENT_RATE_PROFILE,
    ADJUSTMENT_PITCH_RATE,
    ADJUSTMENT_ROLL_RATE,
    ADJUSTMENT_PITCH_P,
    ADJUSTMENT_PITCH_I,
    ADJUSTMENT_PITCH_D,
    ADJUSTMENT_ROLL_P,
    ADJUSTMENT_ROLL_I,
    ADJUSTMENT_ROLL_D,
    ADJUSTMENT_ALT_P,
    ADJUSTMENT_ALT_I,
    ADJUSTMENT_ALT_D,
    ADJUSTMENT_VEL_P,
    ADJUSTMENT_VEL_I,
    ADJUSTMENT_VEL_D,
    ADJUSTMENT_MAG_P,
    ADJUSTMENT_POS_P,
    ADJUSTMENT_POS_I,
    ADJUSTMENT_POSR_P,
    ADJUSTMENT_POSR_I,
    ADJUSTMENT_POSR_D,
    ADJUSTMENT_NAVR_P,
    ADJUSTMENT_NAVR_I,
    ADJUSTMENT_NAVR_D,
    ADJUSTMENT_LEVEL_P,
    ADJUSTMENT_LEVEL_I,
    ADJUSTMENT_LEVEL_D,
```
With the exception of `ADJUSTMENT_RATE_PROFILE` which is already implemented in an 'absolute' way, in the sense that it uses a 3POS switch with each position mapped to a rate profile (1, 2, 3).

@joshuabardwell I like the simplicity of leaving the mapping in the TX. But as you say, it would be hard for users to implement as you would have to set up a complex mixer on the aux channel with the pot in the case where you have different adjustment functions configured via a different 3POS switch. For example, you would have one range of values for P, a different one for I and a different one for D. 
It seems like it would more naturaly to set up the ranges in the configurator since you're already mapping the switches to functions.
Also, good point about the TXs without complex mixer functions.
A first implementation could be without the addition of these custom ranges.

@lostcontrol You have an interesting idea there. One problem I foresee is 'remembering' the saved value. The saved values are read from the EEPROM at boot and in flight adjustments directly modify the value in memory. This solution would require either:
 - An additional variable for each adjustment range that always keeps a copy of the last saved value (increased code complexity)
 - Re-reading from the EEPROM with each loop, though this is clearly a bad idea/unreasonable.
The downside is that you still have no way of having a frame of reference of what your actual values are in flight. If you go through the adjust, save, adjust cycle, you don't have a clear representation of the range of values you're working with is.
Part of the advantage of the 'absolute' method is you always know what your endpoints are and get an intuitive feel for what the numbers of each value mean in terms of flight. It's easy to visualize what your current setting is just by knowing the endpoints and looking at the pot.

lostcontrol, 2016-02-18:
@grokasm I would also prefer the "real absolute" method, I'm completely with you. I was just thinking out loud trying to minimize code changes.

Also I see at least 2 use cases.:

- The  first is e.g. adjusting PID were the method I described could work well since you will likely do it a few times and then de-configure the settings once your copter is tuned (and assign another function to the pot).
- The second is e.g. having a variable RC rate where you want to keep this functionality available all the time and never really want/need to save the value. Here this method would not be so good.

For the "real absolute" method, end points are needed I think and should be defined in the FC.

chasers, 2016-02-18:
+1 what @gpolic said.  The implementation in OpenPilot was pretty decent. 

I think it would be rad to use one pot to pick which thing to act on (via setting ranges) and then use the other pot to actually change the value of that thing.

I think setting min / max values in CF would actually be the most straightforward way to do it.

grokasm, 2016-02-18:
I've never used OpenPilot so I haven't seen their implementation. @chasers @gpolic, could you guys paste in some screenshots of the config UI from OpenPilot if you have access to it?
Also @chasers, I think using a pot to select which thing to act on is within the current functionality of cleanflight for the adjustment functions where you select a channel and range which 'activates' the function/value to adjust. But yeah, agreed, using a pot or a 6 pos switch really opens up the possibilities of how much you can configure in-flight with just two channels.

complacent1, 2016-02-18:
@lostcontrol your second example is the one that I am most interested in. On my Multiwii quads I am constantly adjusting PIDs mid flight to gain better overall flight characteristics.  For example when I want a quick decent I raise my Pitch& Roll P&D to prevent oscillations, then I lower them as I take off and move forward.

This brings another question to the table regarding implementation.  With the Arduino code that I am using with Multiwii I have Pitch and Roll P and D tied together onto pot.  This makes for an amazing overall PID feel that you raise and lower via a single pot. I would love if something like this could be implemented where you can mix more than one adjustment function to a single pot. Works amazing on Multiwii. This could certainly complicate things, but its an amazing way to adjust PIDs on the fly and it changed the way I think about mid flight tuning from "a way to tune" to "a better way to fly", if that makes any sense.

grokasm, 2016-02-18:
@complacent1 I actually think this would be possible if we just added custom ranges to the absolute values. I believe the current implementation allows for up to four slots or functions that can be controlled simultaneously, even with a single channel. 
In the multiwii example, I believe the P and D values are coupled by a factor of 10, or rather, D is set to P/10 (or something close to that). This could be configured in cleanflight by assigning pitch and roll P to the pot channel in 1 slot with a custom range of 30-80 (made up numbers) and pitch and roll D to the same pot channel in another slot with a range of 3-8 (again, made up numbers).
If I'm understanding your use case correctly, it would be completely doable via the configurator with new min and max values added to the GUI (in addition to inside cleanflight itself).
Perhaps we could support ranges that go backwards (min > max) so that an increase in one value on the pot would be a decrease in another value on the pot. But maybe I'm going overboard... :stuck_out_tongue_winking_eye:

complacent1, 2016-02-18:
@grokasm I think you are right so that sounds like it would work just fine.

Reverse adjustments might come in handy, that's a cool idea. For example when I raise P ad D of Pitch and Roll I tend to lower Level P as they somewhat offset each other. Same for opposites, when I raise the PIDs for pitch and roll I tend to drop Level O to compensate so the quad doesn't stiffen up on me too much in Horizon mode.

Very interesting thought indeed.

gpolic, 2016-02-19:
@grokasm , here is some information on LibrePilot Tx PID feature:
https://librepilot.atlassian.net/wiki/display/LPDOC/TxPID

Here some video on yt I just found, Its for Openpilot but its pretty much the same:
https://www.youtube.com/watch?v=VbZNeDCnUyI

Its around 10 months since I used this feature on Openpilot  :-o
You can select which value to fiddle with(PID Options), then select the appropriate AUX channel (Control Source) to affect the value. And then enter the Min, Max.

In the video as you can see the values changes live, and the UI changes values when the pot is moving.
In OP/LibrePilot you can have 2 AUX channels and the throttle channel !! which can affect 3 values at the same time

For me it would be convenient to use one 3way switch and a pot.
The switch may be selecting the variable that is affected (PID). And the pot can be used to alter that value. 

If the user can create a 6 position switch on the Tx (or a mix) then he will be able to change 6 values e.t.c. with his pot.
Considering the small pots that we have, one accidental movement can result in huge change in the value and then boom crash. Of course this depend on the endpoints. This is why I like the Min/Max option.

grokasm, 2016-02-19:
:+1: Thanks @gpolic!

ledvinap, 2016-02-19:
@gpolic:
How is handled switching between controlled values?

grokasm, 2016-02-19:
I've been thinking about this and one of the issues of having various values selectably adjustable via a single pot using the absolute values is with switching adjustment functions.
Perhaps @lostcontrol had this in mind as well and I think this is also what you are asking @ledvinap (correct me if I'm wrong)?
So let's say we have a 3POS switch mapped to selecting Pitch+Roll P, I, and D respectively for the pot AUX channel and we want to use this functionality for initially tuning a new build.
We adjust P to our liking, switch to D and adjust and then move on to I (imagine consecutive switch positions in that order). If we then switch back to P adjustment, the last I value will still be in memory from the pot but the D value will have assumed whatever the pot value is currently during the switching. So we think we're tuning P with I and D to where we set whereas D has momentarily changed during the function selection.
One solution to this scenario could be to have a complex mixer on the tx that uses two switches to map 6 or more values to a single AUX channel. Then one of the switches could control 'enable adjustment' and the other switch could control 'select function'. In cleanflight, the aux range corresponding to the ranges with the enable switch off could be mapped to 'no changes'.

gpolic, 2016-02-19:
Haha yes you're right. If the switch is going up and down then the values will reset to the POT value. It will be a mess!  I havent thought of that!

Well if someone would like to keep it simple then use only POTS for AUX channels. One POT to change only one value.  You can have one or more POTS assigned. 
For example assign one POT to P and one POT to I, and tune. When finished, go to configurator and without disconnecting the battery, save the values.  Then change assignments for D and Rate, then tune again. 

Pretty much that is how Openpilot TxPID does it. They do not do switching between controlled values as far as I know. They assign one controlled value to one AUX channel. The controlled values maybe different than in cleanflight though (not sure)

For the value mapping:
Use mix/max values or the percentage as @lostcontrol suggests. Also very good because it need less code changes.

gpolic, 2016-02-22:
I tried to look at the code... wow thats too complicated :-o
But I found where the changes are happening. Took several hours to understand the data structures and process but cant say that I got it 100%

Trussroads, 2016-02-23:
Are you meaning that instead of using the pot like a 3pos switch for an up and down movement, that for example the pot would be instead given a target item to adjust (eg. Pitch/Roll- 'P' value) and a range of adjustment (5.0 points for example) and then the pot position would directly reflect that P value by position? (Centre detent = +2.5 -2.5)

I agree in theory that to assign a value and have accurate pot adjustments would be great as the way I have my TX switches/pots assigned I do need to use a pot for the adjusting, so I have no clear idea of the amount of adjustment I made until I save and reconnect the board to view where the value actually ended up after the flight? Sometimes this reveals huge adjustments and why the response in flight was poor? Perhaps the adjustment speeds up the further it is turned? I am not sure on the exact function? It may be why a switch is recommended instead? Can someone clarify?

Also wondering what the LED flash represents when an adjustment is made and if it represents a specific value of adjustment each flash?

gpolic, 2016-02-23:
With the current config on the Adjustments, you can have one switch to select the affected value (P,I , D .e.t.c.) . And another switch to change the value up or down.
Also you can skip that, and assign only one switch(!) that increments or decrements one specific value (only one)

This is increasing or decreasing the value selected, _over time_. 
The more time you have your switch UP, the more the value is increasing.

The preference is to change the value according to the POT position. Not over time, but on the POT value.

Now how that is going to happen, this is been discussed here.
1. Moving the POT up or down, can transfer the "absolute" channel level to the value
2. Or, it can increase and decrease "relatively" around a predefined center value (saved PID on the EPROM)

Of course there must be some safety on the range of values(endpoints), this it can be either configured Min, Max values, or a configured percentage. 
When we add "configured" values, I believe that there are changes to the CF gui and the CLI. (more work needed!)

Hope that is clear and I hope I got it right !
I don't know what the change of the LED means, sorry.

Going back to the previous discussion On switching channel(3 POS?)  and POT.
I thought about 2 stuff on the switching.
One is to introduce a delay when the switch changes the affected value, before taking the POTS readings at all. This will help the user to first stabilize and make sure he got the correct switch position. Also by switching up and down the values will not be affected until the user settle the switch on one position.

Second. 
Switch to your channel with AUX1 and then start tuning with POTS.
How about to wait until the POTS is in midrc i.e. 1500 (approx)  before taking the POTS value to the configuration.
That means that the user may fiddle his 3 POS switch all day, but _only_ when the POT is in the middle it will start to "read" values.

Just trying to find a solution to what @grokasm said. If you are going to a totally different direction please let me know and I will stop mumbling.

I am still in favor of tuning with 2 POTS as in the AlienWii example on post #1  :-)

lostcontrol, 2016-02-23:
Sorry to mention APM:Copter again :stuck_out_tongue_closed_eyes: but there you can have one single channel used for pot adjustment (maybe this changed in the last versions). This means you say I want to adjust this value with the pot. If you want to change what you adjust, you have to reconfigure it in the FC. Maybe this pot adjustment feature should be separated from the in-flight adjustment feature!?

grokasm, 2016-02-23:
@lostcontrol I like augmenting the in-flight adjustment feature because it is so configurable. The end user can set it up as they wish. They can have one pot per function that is fixed, or they can configure switchable function on a single pot.
My concern about the switching would only present itself in certain configurations which aren't by any means the only use cases.
@gpolic I don't like the predefined center approach because it seems needlessly complex in dealing with keeping an updated copy of each value saved in eeprom in memory and we lose the simplicity of min-max to pot range mapping.
Your idea about having a timed delay is interesting and there is already a timing mechanism in place for the switch-based adjustment which could be adapted for something similar.
I would implement a first version without this delay and see how it goes, then, if there is interest, we can open a separate issue for such improvements. Thanks for the suggestion!

gpolic, 2016-02-23:
ok sure @grokasm can you please describe how the first version is implemented ?
With all the items that are discussed i am confused !

Trussroads, 2016-02-24:
Yes i understand that and has clarified why I sometimes have very large adjustment unexpectedly when using a pot.. (Because of time based change) Thanks for explaining.

I think step one could be a change only on the adjustment ability of the pot position from time based, to instead make the pot position represent an actual value as you suggest and relative to its place in a predetermined range of the total sweep This would need to be configured for each value I'm thinking? (low=X <-> high=X) 

Once this is configured then the next step would be function or allocation is then added to the selection and which value is appropriate for ranges on each item. This may be different on each PID controller too actually and could complicate sensitivity of adjustments? I'm not familiar with code limitations across multiple PID's?

Regards,
Scott..

Yamaford, 2016-02-25:
Could the POT be set up like we fly in rate mode? turn the knob left to go down on P. The further you turn it, the  further the P goes down. When you turn right from full left your P doesn't start climbing until you reach center. This would be more intuitive for rate mode flyers and act as a ratchet. 
As it is now on a 3 pos switch you can go down one point at a time center down center down center down. The way I described the knob could drive the P down say 10 points at a time. Center full left center full left center full left giving you 30 point drop. Clear as mud huh?

reedbn, 2016-05-15:
Pretty sure this is the same as Issue #10

closed 2017-02-25 by hydra