cleanflight/cleanflight #1843: GPS modes broken in V12 RC2

OleEngr, 2016-02-18:
I first tried a GPS with my ZMR quad with V12 RC1.  With some PID adjustments for GPS, it worked beautifully.  I could send the quad downrange 50 yds or more and engage RTH.  It would make a 180 turn and fly back close to the starting point.  when it overshot, it would do a 180 turn and return.  GPS Hold also worked very well, with about the right amount of control to keep it  reasonably stable.  
With RC2 and the revised configurator, it is essentially non-working.   When Hold or RTH is engaged, it starts to navigate away from the direction it should be taking, and it manual control has to be applied to keep it from crashing.  I tried to restore to RC1, but I have not been able to get anything remotely close to what I had before, and I have a backup and CLI saves.  
I don't think anything has broken on the Quad.  The GPS operation as shown in CFC seems the same, and a GPS fix is always signaled quickly.  There was a configurator change that occurred when RC2 was installed, for what it's worth.  
I always tested the GPS modes with ALT HOLD engaged, but that also seems worse in RC2.

skaman82, 2016-02-22:
Do you really used CF 1.12 RC1? The 180 turn on RTH sounds like iNAV branch. As far as I know, CF don't have a 180° turn on RTH.

OleEngr, 2016-02-23:
I'm not dreaming or hallucinating.  The quad made a turn from its original track so it aligned with the Return Nav bearing (I'm assuming).  It was really neat and it was definitely RC1.  I wouldn't know how to find other firmware, other than what is accessible by following the link from Cleanflight.com.  Maybe that explains why I could not get the same results when I tried to restore to RC1.  
 
I tried V12 (released), but had problems with the quad not arming unless I stripped most things out of the mode selection.  I went back to RC1 for the second time and again found that GPS didn't work as before.   I would have prepared an issue, but it was too bizarre and I thought maybe it was something I did, although there definitely seemed to be something in the config that prevented arming, and it was able to arm when I removed enough stuff from the configuration.  
 
Doug
 
Date: Mon, 22 Feb 2016 15:48:29 -0800

OleEngr, 2016-03-02:
I recently found out that the mag compass in the quad with GPS seems to be defective.  It will no longer statically report a heading in the configurator.  Unless somebody else has the same issue, it was likely that the initial problem I observed with V12 RC2 was either a PID problem due to the changes in RC2 or the compass was damaged between the last time I used it with RC1 and the installation of RC2.

hydra, 2016-03-09:
ok, so nothing to fix then.  closing.  GPS features will improve when the inav code is merged.

closed 2016-03-09 by hydra