cleanflight/cleanflight #1848: Return temp (gyro or baro) on smart telemetry

brycedjohnson, 2016-02-18:
I have an enclosed waterproof quad and I would like to keep track of temp to make stuff isn't getting too hot.

Currently cleanflight is sending a mode flag over smart telemetry for temp1

https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L400

And Gps lock status for Temp2

https://github.com/cleanflight/cleanflight/blob/master/src/main/telemetry/smartport.c#L440

Can we send back the data like frsky.c?

https://github.com/cleanflight/cleanflight/blob/64797d164861011cab4ea8449cc334bfbeb88424/src/main/telemetry/frsky.c#L209

I can make the changes either by removing the mode flags on temp1 or adding an extra else if (if no gps) on temp2.

Anyone have opinions?

3djc, 2016-02-18:
I would not remove the flags (I find it VERY usefull to be certain on what mode the FC actually uses),  In fact, I was planning to make a propoal to make them available in older telemetry too (to bring them more in sync) as part of a bigger review of that area, But I think your idea to send actual temp with tmp2 when no GPS is present is great !

brycedjohnson, 2016-02-18:
Sure that makes sense.  I saw the inconsistency between the telemetry options and I wasn't sure if it was being used.  I'll throw something together for temp2 if GPS is not being used.

hydra, 2016-02-20:
does it not make more sense to send the temp back in the correct place and change the opentx code and add support in cleanflight for the various mode/flags/states?

I vaguely remember someone else wanting changes to the mode/flags/states telemetry support so it might be worth searching the issue tracker for opentx/frsky/telemetry and see what issues and PR's might also be relevant to this discussion.

closed 2017-02-25 by hydra