cleanflight/cleanflight #1851: SP Racing F3 and FlySky i6 - Receiver mapping issues

fast240z, 2016-02-19:
Receiver mapping appears to be an issue with Cleanflight/SP Racing F3/Flysky ia6 when connected via PWM.  

Channels are mapped correctly in Cleanflight and are connected to the appropriate pins on the rx, but Yaw and Throttle indicators move at the exact same time and rate when the throttle is moved.  Applying Yaw via the sticks does nothing.  The same behavior happens with Roll and Pitch, both roll and pitch move at the same time when roll is applied.  Applying pitch does nothing via the sticks.

I've switched the pinout and found that the other channels do function, but for whatever reason the roll/pitch and yaw/throttle values are "locked" together.  

Google searches indicate this has been a problem on Naze32 boards in the past, but no one has actually provided a solution that I've seen.

hydra, 2016-02-20:
Where did you purchase your SPRacingF3 board?  Shop and Link please.

fast240z, 2016-02-20:
I bought the board from an Amazon seller - http://www.amazon.com/gp/product/B018VQFWU2

Wiring photos are attached.  I followed the wiring guide in Painless360's video.  The channels seem to have paired up, 1-2, 3-4, 5-6 all move at the same time.  I have no mixing set on my radio, and I've reset it back to factory settings with no luck.

[image]
[image]
[image]
[image]
[image]
[image]

# dump

# version
# Cleanflight/SPRACINGF3 1.12.0 Feb 19 2016 / 20:01:10 (aa8608a)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature VBAT
feature FAILSAFE
feature RX_PARALLEL_PWM
feature BLACKBOX

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
[... 278 lines cut ...]
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

jlj0425, 2016-03-15:
i have the same issue. before i use 1.11.0 all is working well. when i flash it with the fucking new 1.12.0 Yaw and Throttle indicators move at the exact same time and rate when the throttle is moved. Applying Yaw via the sticks does nothing. The same behavior happens with Roll and Pitch, both roll and pitch move at the same time when roll is applied. Applying pitch does nothing via the sticks. tried to re-flash using 1.10.0 and 1.11.0 again to 1.12.0 again and again still it does not work. i also tried to brick my F3 and boot again but still not fucking working.  cant use my quad anymore. shit

sammykumson, 2016-03-16:
i am having the EXACT same problem ...PWM NOT WORKING ON SP F3 DELUXE. AR610 RECIEVER DX6 all works fine on a Naze32... and others ... i know its not wired wrong or mapping ... i setup many of the Naze32 ... Cleanflight 1.2.1 ...... I will Not go to ppm as i have all Spektrum gear. can some one please fix or tell me how to as i would really like to use this Flight Controller as it all wired up ..... bought it from ReadytoFlyQuads.com  ...

StraTech, 2016-03-28:
I am having a similar problem on SP RacingF3 & FS-IA6 I can't get any input on CH1-CH4, but AUX CH5-CH8 working something is mapped wrong have tried v1.10.0 v1.11.0 v1.12.0 all the same. I can get ppm working with another FS-IA6B RX but I still need PWM working!

jlj0425, 2016-04-01:
Good news Ihave just found the solution. it doesn't give me any more head ache.. i just smash my fucking F3 with a hammer. after then no more head ache. LOL.

hydra, 2017-10-21:
use v2.0.0, if the problem persists open a new issue and reference this one.

closed 2017-10-21 by hydra