cleanflight/cleanflight #1859: Cant use CF on my CC3D Atom

lenguyendan, 2016-02-21:
I have flashed the CF v.1.10 to my CC3D Atom but, i cant connect it with CF configurator 1.2.0.
I have  my blue led flashing slowly on my board! please help me to solve this.

Quajetson, 2016-02-23:
In GCS run Rescue procedure and select CF firmware 1.10 in BIN format. When done click SAFEBOOT buton and disconect board. DONE:) Now if you have FTDI you can upgrade to latest form CLEANFLIGHT firmware tab in GUI

gregm412, 2016-02-23:
How do you connect with an ftdi to cleanflight?  And do I just choose the latest cc3d

Quajetson, 2016-02-24:
im using mainport it's normally detected in Configurator

sppnk, 2016-03-17:
@Quajetson thank you very much. After many many days trying to flash my CC3D Atom boards through STM Flash demonstrator method, I have had SUCCESS just from CF configurator !!!

Method I have used for CC3D ATOM:

1.- Short BL small pads with small screwdriver.
2.- Connect USB to power it. The green led lights on, and nothing else.
3.- Connect FTDI to PC and see the corresponding COM showing itself in CF configurator (you have to install FTDI drivers first).
4.- Flash it from CF and go to fly !!

What I dont understand is why it didnt work with stm flash demonstrator..... :-(

rdpeake, 2016-03-17:
What i have found with my CC3D Atom, if I have anything connected to the main port during power on, the cc3d enters bootloader mode. Thus i am able to flash the cc3d using the flash on connect feature of CF configurator... makes it annoying to use sbus on main though.

Quajetson, 2016-03-17:
@rdpeake - This is normal situation when you have already Cleanflight onboard.
But when you have stock CC3D with openpilot is not possible.

Quajetson, 2016-03-17:
@sppnk  STM flash works.
For me STM works always for second try. At first try i get connection error.
When this message is on screen. I'm disconnecting board and connecting it with SBL & +3v bridged for moment
(when bridging OK CC3d has a ORANGE led on) Then unbridging and reconnecting without closing STM and getting READY:)

rdpeake, 2016-03-17:
@Quajetson Are you refering to the main port causing bootloader mode, or just the ability to update firmware?

If it is bootloader mode, is there a way to disable it, as i have my sbus receiver connected, but must remove it when i power on, otherwise i don't get successful boot?

Quajetson, 2016-03-18:
This is not booloader mode. Bootloader is only when SBL +3V is bridged and
ORANGE led is ON.
Cleanflight update from configurator is possible only when Cleanflight is
on already board.

rdpeake, 2016-03-18:
When I have something plugged into the main port in cc3d atom, I do end up with just the orange light and no USB port functionality. Is this caused by cleanflight, and should I be starting a new issue about this to have it looked at? Or is it hardware?

Quajetson, 2016-03-18:
No USB port functionality? Unknown device in Windows?
Tray at this moment use FTDI + STMFlashLoader Demo
For me when CC3d is in bootloader mode is not recognized by windows.
Only FTDI connection is possible at this moment.

rdpeake, 2016-03-19:
I am running 1.12.

If i have my receiver (sbus) plugged into main when power is applied to the 5v line, or even if i just have a cable plugged into main when i power the board, it always fails to boot.

I can get it to boot just fine, if i either, power the FC via usb prior to powering the 5v rail, or by removing the cable connected to main.

Soft resets, such as save and reboot, have a chance of not rebooting correctly as well, though, hot starts seem to fail less that cold starts.

Quajetson, 2016-03-19:
Try with 1.10 and tell what is on this version, but probably your board is
broken. With openpilot boot normally?

rdpeake, 2016-03-19:
I've not used openpilot, so not sure there. However, i just flashed a new cc3d atom with cleanflight 1.10, default settings, and the problem still occurs. If i have the main port cable connected to my receiver, there is no boot. If i have the main port disconnected, it boots.

I can also get it to boot if it is powered via usb first.

Quajetson, 2016-03-19:
If you try boot with SBL + 3V bridged whats happened?

rdpeake, 2016-03-20:
I attempted at one point to boot with sbl bridged when i was trying to update cleanflight, didn't seem like it worked, though, you've since pointed out that i don't need to do this to update cleanflight if it is already installed.

Do you mean for me to do this in place -> something plugged into main, or just in general?

Quajetson, 2016-03-21:
I want to known how your board boot with this bridge.
If you don't want openpilot try Betaflight maybe your problem  solve itself
there are some differences between these softs.

rdpeake, 2016-03-21:
Booting with sbl +3v results in the same functionality as booting with something plugged into serial.

for reference i am using this board: http://www.hobbyking.com/hobbyking/store/__75572__CC3D_Atom_OpenPilot_Flight_Control_Board_Authorized_Vertical_Pins_.html

I am thinking this is likely a hardware configuration problem.

closed 2017-02-25 by hydra

joshpaulcloud, 2017-04-12:
I've tried all this and it does not work. I'm on Ubuntu and I my user has permission to the ports and I have modemmanager off. 

Betaflight configurator DOES recognize the board when I flash it back to Librepilot firmware. It also recognizes a separate CC3D and a Revo on Librepilot firmware. But it does not open the port when in boot mode with or without the cleanflight_CC3D.bin file.