cleanflight/cleanflight #1865: Failed to open serial port

olaf2273, 2016-02-21:
hello
i cant connect any more  to clean flight.
now I have this in menu:
2016-02-21 @ 17:10:11 -- Failed to open serial port
2016-02-21 @ 17:10:22 -- Serial port successfully closed
2016-02-21 @ 17:10:25 -- Serial port successfully opened with ID: 2
2016-02-21 @ 17:10:35 -- No configuration received within 10 seconds, communication failed
I booed the naze32 but no response  what to do.
it workt before

SteveAmor, 2016-02-21:
Suggest you look for support on forums such as RC Groups and IRC #cleanflight. 
Please close this issue. It is not a bug

olaf2273, 2016-02-21:
before the old clean flight program it works. and now clean flight is updated its doesn't works ?

Arti3DPlayer, 2016-05-14:
have same problem. how to solve it? Write guide please...

olaf2273, 2016-05-14:
I bought a new One,
At banggood. A new one naze32 V6
This wan js wurking good !!

http://m.banggood.com/CK-REV6-NAZE32-10DF-Flight-Controller-For-RC-Multirotors-Baseflight-Configurator-p-1030040.html

olaf2273, 2016-05-14:
And its surk with glean flight

SteveAmor, 2016-05-14:
Hi @olaf2273 good to hear you fixed the problem. Please can you close this issue. Thanks.

closed 2016-05-15 by olaf2273