cleanflight/cleanflight #1867: Graupner HOTT Telemetry not working

ebawho, 2016-02-22:
I previously had graupner hott telemetry working on cleanflight, in version 1.7. I am however unable to get it to work in the latest version. I would like to run the latest version for obvious reasons and would appreciate any help in directing me in how to enable the graupner telemetry.

Yodero, 2016-02-28:
Have the same problem and used a logic analyzer to trace this problem. At selection frsky, I can see data coming from softserial2 on my Spracingf3.
After switching over to hott, nothing is transmitted anymore.

ebawho, 2016-02-28:
I got it working by setting both soft serial 1 and 2 to hott.

Yodero, 2016-02-28:
Thanks! I trief that as well, but it didn't work for me. I got it to work 2 days ago, but after I changed some settings the day after, telemetry died :D. Becazse I am familiar with coding and microcontroller, I put my logic analyzer at the softserial wire. Nothing happend, until I switched over to frsky! I can read all data for frsky telemetry...

ebawho, 2016-02-28:
Yes, I've had mine seemingly randomly stop working as well. Had to ref lash
and reconfigure multiple times (even loading known working saved config or
cli dump wouldn't fix) the only time I had predictable and consistent
graupner telemetry (even had menus to adjust settings) was a custom version
of clean flight, but the latest version I have of that was based on 1.9
unfortunately.

Yodero, 2016-02-29:
Maybe they find some strange condition, that prevents telemetry from working. If I have a bit of free time today, I'll look into the code - maybe I can find something obvious.
Hope they'll fix it in the next version!

Pierre-A, 2016-03-19:
**@ebawho** , **@Yodero** , sorry for the delay... did you get Hott telemetry working on your SPRF3 ?
as it works fine on mine.

Yodero, 2016-03-21:
@Pierre-A , I didn't try it since then, but my Graupner doesn't show any telemetry information.
As I mentioned, I thought I maybe have an issue with my wiring and put my logic analyzer directly to the ports. After testing around, I get a signal when using frsky telemwtry, but not when using hott.

I will post my setting tomorrow, maybe you can see something strange?

Regards
Yodero

Pierre-A, 2016-03-23:
**@Yodero** no setting ? thanks if you have time, just in case it would be bug related.

Yodero, 2016-03-23:
Hi @Pierre-A. Sorry, I'm a bit late :)
Here is a dump of my settings.
[dump_23.03.2016.txt](https://github.com/cleanflight/cleanflight/files/186764/dump_23.03.2016.txt)

Hope I don't have some misconfigurations... :)

Pierre-A, 2016-03-24:
**@Yodero**, I tried your dump on my SPRF3 with a GR12L (on the bench, no GPS, no Blackbox) and Hott telemetry is fine.
What is your RX ? can you re-check it's configuration ? which signal on the RX telemetry pin ?

johnlenfr, 2016-04-01:
Hi Pierre, seams to have an invertion on Softserial with the SP Racing F3.
But I doesn't have the time to explain in detail.

Don't know if it is a problem in GUI or in the SP Racing F3 hex file but to have Telemetry on IO-1 port I have to select Hott telemetry on Softserial 2 in the GUI.

See issue #1745

Chears,
John

johnlenfr, 2016-04-01:
@Yodero : why is your telemetry_inversion = ON ?
Default is OFF and I did not change this parameter to have the Hott Telemetry be functionnal.

Try to put 
set telemetry_switch = OFF
set telemetry_inversion = OFF

And test 

Also verify your connection is ok with your diode: https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md#hott-telemetry

Pierre-A, 2016-04-01:
**@johnlenfr**, hi John, I don't understand this issue as the config works as expected for me with a GR12L on my SPRF3 : Hott on Softserial2 and telemetry on IO2 Ch3/Ch4.

closed 2017-02-25 by hydra