cleanflight/cleanflight #1868: SONAR pins are not correctly detected in pwmInit()

jflyper, 2016-02-22:
SONAR pins are not correctly detected in `drivers/pwm_mapping.c:pwmInit(drv_pwm_config_t *init)` because the `init->sonarGPIOConfig` in  points to invalid location.

This is caused by the target `sonarGPIOConfig_t sonarGPIOConfig` is declared inside a local scope for `if (feature(FEATURE_SONAR))` code in `src/main/main.c:init()`, i.e.:

```
if (feature(FEATURE_SONAR)) {
        sonarHardware = sonarGetHardwareConfiguration(&masterConfig.batteryConfig);
        sonarGPIOConfig_t sonarGPIOConfig = {
            .gpio = SONAR_GPIO,
            .triggerPin = sonarHardware->echo_pin,
            .echoPin = sonarHardware->trigger_pin,
        };
        pwm_params.sonarGPIOConfig = &sonarGPIOConfig;
}
...
pwmIOConfiguration_t *pwmIOConfiguration = pwmInit(&pwm_params);
...
```
The sonarGPIOConfig should be declared so that it is valid when the `pwmInit()` is called. Quick temporary hack would be:
```
sonarGPIOConfig_t sonarGPIOConfig;
if (feature(FEATURE_SONAR)) {
        sonarHardware = sonarGetHardwareConfiguration(&masterConfig.batteryConfig);
        sonarGPIOConfig.gpio = SONAR_GPIO;
        sonarGPIOConfig.triggerPin = sonarHardware->echo_pin;
        sonarGPIOConfig.echoPin = sonarHardware->trigger_pin;
        pwm_params.sonarGPIOConfig = &sonarGPIOConfig;
}
...
pwmIOConfiguration_t *pwmIOConfiguration = pwmInit(&pwm_params);
...
```
I mean it's temporary because there are targets (e.g., sparkey2) which uses different gpio pins for trigger and echo, which can not be resolved in the current logic; we need separate gpio member for trigger and echo. I believe the whole sonar detection code should be re-written to make it sound. We should also note the issue #1860 when we do so.

martinbudden, 2016-02-23:
I've updated pull request https://github.com/cleanflight/cleanflight/pull/1861 to include your `sonarGPIOConfig` scope fix.

I agree that the passing of parameters to `pwmInit` needs to be looked at and fixed.

hydra, 2016-08-23:
PR merged.

closed 2016-08-23 by hydra