cleanflight/cleanflight #1870: Cycletime is not accurate in the current code

borisbstyle, 2016-02-22:
In the current code cycleTime is based on the frozen value before taskMainPidChecker.
https://github.com/cleanflight/cleanflight/blob/master/src/main/mw.c#L626

There might be a significant difference in the real cycletime especially with gyro_sync enabled. The worst possible scenario is when a board doesnt have INT pin connected it would show 1000us, while it is actuallty 1100us.

ledvinap, 2016-02-22:
cycleTime should be related to data sampling, not processing. With INT pin working , it will be very close (off by gyro clock frequency deviation).
Without INT pin, the 9 samples will be 1ms apart, 10th sample will be discarded and 11th sample used next (with 2ms delay). Using cycleTime of 1.1ms won't help much ...

borisbstyle, 2016-02-22:
@ledvinap 
I am not talking about processing, but I dont feel its a good thing to "lie" about the cycletime :D

borisbstyle, 2016-02-22:
I also played with different appriaches than the guard interval in current scheduler. I think that implementing a buffer period inside GYRO/PID or just motor write gives better resutls than reserving guard itnerval in scheduler.

ledvinap, 2016-02-22:
As I understand it, cycleTime and dT is used data timing and for filter delta time. Using value from gyro sampling will IMO get best results.

My feeling is that from physical point of view jitter under 0.5 ms wont have noticeable (possibly even measurable) effect on fight performance. But there may be some aliasing/interference in ESC firmware that will amplify jitter/timing influence (randomizing jitter may theoretically improve performance).

It would be great to implement some experiment that will measure/quantify jitter  and delay influence on performance.

borisbstyle, 2016-02-22:
targetLooptime is used in all calculations in the current code. No jitter introduced there. You always get the sampling frequency.

The cycletime is just a visual representation in configurator and very usefull for jitter monitoring. And that is giving a false indication right now!

I know that jitter definately impacts the performance of ESC firmwares......only not till what level. It can also make calibration of ESC difficult on some firmwares.

Just a little time buffer in gyro/pid task will completaly solve the jitter. But it does depend on hardware how long it has to be

ledvinap, 2016-02-22:
You are right. Local dT did confuse me .. 

As for the jitter - you need buffer longer that maximum jitter time. But it is reasonable to restrict all tasks to some maximum time (task has to be split if it can take longer). Jiter will than be tak time + gyro processing delay, which should be bounded.

closed 2017-02-25 by hydra