cleanflight/cleanflight #1873: Problem with 1.12.0 and 1.12.0 RC2

ghostsharp, 2016-02-22:
Using:
Naze32 + RX_Serial + (SOFTSERIAL) GPS Ublox + Telemetry (smartport): ARM but dont start motors

Changing to 1.12.0 RC1: All work

I dont know which is the difference betwen versions.

3djc, 2016-02-22:
Try disabling gyro_sync,  or try gyro_sync_denom set to 2

ghostsharp, 2016-02-22:
Solved with gyro_sync_denom = 2

3djc, 2016-02-22:
Great ! Could you close this issue then?

tmenet, 2016-02-22:
I was having a problem with gps and luxfloat enabled at the same time but not with multiwii rewrite. It would not arm in Luxfloat mode. I set gyro_sync_denom set = 2 fixed it for me too. Still havng a problem getting g-tune to change pid's. Perhaps I am still over taxing the F1. I will try to get more data.

skaman82, 2016-02-22:
Had the same problem with arming, but gave up before reading this and moved back to 10.11. 

@hydra I think, if people having problems with different settings or additional hardware (gps, ext. baro or compass) the default settings of current CF are not optimal. It should work for everyone and each (or at least most) configurations - F1 targets are still very popular. Please think about turning off gyro sync as a default or gyro_sync_denom set = 2 as default. Maybe this requires some other code optimization. The defaults should not be focused so much on performance and overclocking. It should be all about compatibility. Sometimes I have the feeling the development is all about ACRO mode and racing - all leveled flightmodes aren’t getting enough love and testing. The performance in leveled flight modes is getting poorer (at least in some of my tests - see drifting issue) and now we have problems if we want do more with CF than racing. All this makes CF much less attractive for beginners - at least I think so.

tmenet, 2016-02-23:
@hydra  I agree with @skaman82 's comment above. Or perhaps maybe your intention was to have the default gyro_sync OFF like it is written in the [cli document](https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md) but in fact the default inadvertently comes preset to ON.

I have 3 quads with Cleanflight, A 3D printed  MHQ2 to experiment with GPS and new features. An indoor micro quad. And a ZMR250 I am building for racing. I agree that the defaults It should be all about compatibility. We will find all the performance tweaks for racing.

3djc, 2016-02-23:
I think I was the first one to raise the issue (https://github.com/cleanflight/cleanflight/issues/1794), yet, my thinking has evolved a bit since. 

If you are the kind of person who moves to Luxfloat, adds a GPS, enables telemetry, first of all you are a lot like me, but also and more seriously, you probably have what it takes to make the required adjustement to keep the FC working nicely. Now granted, there should have been much clearer warnings and instructions about that, I'll try to write some to propose to the CF team.

On the other end, I agree with you that there needs to be a place for CF, I hope that this one will be (stay ?) for more casual fliers, when betaflight is more for the 'bleeding edge' pilots.

When I moved to 1.12RC2 on my NAZE32 quad, I did face the issue, but also saw the benefits of gyro sync, even for my low flying skills, and decided two things : 
- set the denominator to 2 but keep gyro_sync activated
- get an F3 FC, and boy, those are VERY nice, I'm now having smartport telem, GPS, luxfloat, all that with gyro sync and denom to 1.

Difficult subject I could not agree more, but hopefully some good will come out for everyone :)

Ravid824, 2016-02-25:
Same here, I am really "not happy" from the priorities set for the CF.  I have the Naze32 and the SPRacingF3. One is used for racing and the other for photography. It looks like that everything done is bias toward the SPRacingF3 which is a wrong direction of keeping a common compatible source for the entire supported quad family.
Not only leveled flight modes aren’t getting enough love and testing, same with GPS functionalities, Altitude Hold and many more which are not functioning well for long time.

Unfortunately, I switched my photography drone to the Naza V2 (not Naze) and have a great stable performance. I wish and want to believe that one day the CF will reach this level.

OleEngr, 2016-02-26:
Much of the comments above are things I have thought, but was reluctant to put in print because of my inexperience and relative short time involved in RC.  My interests in quads include about everything BUT racing, given my age and the reaction time that comes with it, although I have 2 racers just because of their availability and cost.  I looked forward to the other capability that is available with the Naze 32, and the ability to tinker with the guts of it.  Unfortunately, I was not prepared for the steep learning curve it can present.  Between the lack of documentation, the poor style and the ambiguity of SOME of it, I find a level of frustration that is discouraging.  
It's not that CleanFlight is unique, I've seen other highly technical projects, with VERY talented contributors that are narrowly focused.  It makes it tough for those that don't speak in acronyms or think in Hex get onboard.  There have times I have been tempted to put my brushless quads on a high shelf and return to the toy quads where the Flying-time vs Frustration ratio is much higher.  My mother never raised a quitter, so I'll hang in there and look for changes, while hoping that my learning curve flattens out.  
On the positive side, if there was one thing that could help the CleanFlight documentation process, it would be a review period prior to release, where other eyes could make improvements that were missed by the original author.  Years ago, I learned that if there were questions about the technical content of anything I wrote, it needed to be changed so the target audience had questions answered, not raised.

donjakobo, 2016-02-29:
I was also surprised that moving from 1.11 -> 1.12 broke functionality for someone using a Naze32.  I couldn't arm the damn thing, and it drove me crazy for 3 hours as I tried to troubleshoot everything (on a quad that was working on 1.11 just fine, same pids, etc;).

Now I know there is that warning of "don't blindly import backup settings and expect everything to work". But what you effectively say is nothing is going to work, might as well NOT make a backup at all for firmware upgrade routes.

I went back to 1.11 as I was too frustrated with 1.12 not letting me ARM my quad. Still not sure what the cause was (reading things about gyro sync? etc;)

OleEngr, 2016-02-29:
There is a similar topic in the item above labeled: naze32 v1.12 , GPS 3D fix + arming problem.  
The solution is so set the CLI parameter Gyro_Sync_Denom  = 2 (not sure if name is completely accurate).
In my case it armed for a short while upon startup, but maybe when the GPS acquired a fix, it quit arming.  Anyway,  the above setting fixed the arming issue.

dan557, 2016-02-29:
Setting gyro_sync_denom = 2
Fixed the issue for me to

closed 2017-02-25 by hydra