cleanflight/cleanflight #1876: SP Racing F3 voltage telemetry SmartPort.

silverf1re, 2016-02-22:
Maybe this is not possible but I am not able to pass voltage telemetry from my SP Racing F3 through SmartPort to my Tranis x9d. However other telemetry transmits down to my x9d fine such as heading. When the FC is connected to my computer I am able to see my battery voltage in the chrome app just fine.

3djc, 2016-02-22:
Strange, it does work for me? What version of OpenTx are you running on the radio ?

silverf1re, 2016-02-22:
2.0.19 should I be on the 2.1 branch?

3djc, 2016-02-22:
I cannot test 2.0 branch (Taranis X9E) but it should work if you have vbat enabled. You should get it with VFAS

silverf1re, 2016-02-22:
Ill update to 2.1 branch tonight and report back. I appreciate the help and a donation is on the way.

3djc, 2016-02-23:
Please remember that 2.1 requires you to manualy detects your sensors at least once, and you will get a list of the active sensors. If you have GPS, you also need to do a discovery while the GPS is locked, so it will add GPS related data

DzikuVx, 2016-02-25:
I had the same problem. From unknown reason, voltage is sent only when board is Armed. So, do a sensor discovery with Armed board

hydra, 2016-02-25:
this is documented.

closed 2016-02-25 by hydra