cleanflight/cleanflight #1878: BIN Firmware for CC3d?

matzZz, 2016-02-22:
Hi,
is there a *.BIN firmware version (1.12.0) for CC3D?

borisbstyle, 2016-02-22:
Nope.....not supported anymore. No space

Quajetson, 2016-02-23:
Option 1
Install 1.10 as BIN from GCS and update from Cleanflight Firmware tab to latest
Not required bridging 3.3v and sbl pads and external powering
Required FTDI adapter  connected to mainport

Option 2 (single image mode)
You must using hex file when CC3D is connected with FTDI to mainport and (vcc disconected) - powering board from Servo output port. Pads 3.3v and SBL bridged in moment of powering.
Flashing with STM flashing tool 2.5

gpolic, 2016-02-23:
I flash my CC3D with FTDI on the main port, using the CF Gui
If I bridge the pads and give power, afterwards my board is not recognized.
Maybe mine was a bit damaged from all the test its gone through, but I am using FTDI and CF Gui, no problem there

Quajetson, 2016-02-23:
You must powering board from external source. Im using ubec 12v5v and powering board from SERVO OUTPUT PORT np 6. VCC on FTDI must be disconected. Sometimes i must try over 5 times before STM Downloader connect corectly. If you will have problems conntact me i can record video for You.

Quajetson, 2016-02-23:
Here is procedure and soft:
https://github.com/cleanflight/cleanflight/pull/1618/files?diff=split&short_path=e2eb57a

gpolic, 2016-02-24:
Thanks man. I have tried to power the board from external source, short the pads, and then connect with FTDI. 
I tried both STM Demonstrator and the STM32 Flash utility, and no luck. Because everything is working and i got alot of time on it already, i will leave it as it is.
My success in shorting the pads is 1 out of 10 (in a good day!) and I done it so many times that cant fiddle with that any more

As said, I can flash HEX files normally from CF Gui, so the BIN file is not needed for me.
But you must have the FTDI cable to do that.

closed 2016-02-25 by hydra

brandonhill, 2017-09-12:
Sorry to post on a closed issue, but how do I create a binary file myself?

Reason: I have a CC3D without main port, so can only use USB via GCS to flash, thus need a binary. Currently using bin from 2.9.1, but Angle and Horizon modes are unusable, so would like to create a binary myself from latest version. Any help would be greatly appreciated :)

mikeller, 2017-09-12:
@gpolic: Have a look at https://github.com/betaflight/docker-betaflight-build . It's for Betaflight, but should work on a cleanflight repo as well