cleanflight/cleanflight #1879: Enable BLHeli Passthrough over VCP

RC-Slater, 2016-02-22:
Maybe it's just me, but I find that a big sticking point for getting on of the VCP enabled flight controllers like SPF3 Mini, Lux FC, Cyclone FC, and others, is that you can't flash BLheli through the USB port like you can with the regular SPF3, Dodo FC, Naze, etc...

Is this important to others as well? 

 I'm not a programmer, but I'll volunteer my efforts in any way I can.

AndersHoglund, 2016-02-23:
Passthrough should not be (hard coded) connected to neither UARTs nor VCP. It should follow the port mapping of MSP.

Surgikill, 2016-02-29:
I'm having this issue as well. Cannot flash my esc's and therefore cannot fly seeing as half of them are damped light and the other half are not.

timdonovanuk, 2016-03-15:
Splashed out on upgrading from a Naze rev 5 to an SP Mini to find it has less convenient features :/

closed 2017-02-25 by hydra