cleanflight/cleanflight #1883: Disabled Features

eddietheperson, 2016-02-23:
I mentioned this in conversation and it was suggested I submit to issues. It would be really handy if options/features wouldn't show up in the config (or 'greyed out') if your hardware doesn't support the feature. For instance, I modded my zmr into a hexacopter only to blindly find out hex/oct isn't supported in Cleanflight on the cc3d. I know many will say to read the documentation but I feel like good software guides you in a way that for things like this, you shouldn't need to. I apologize if this is the most insignificant issue ever reported. 

Cleanflight is awesome, I appriciate everyone's hard work, you guys/gals are doing amazing things!

hydra, 2016-02-25:
Maybe after #884 is done.

hydra, 2016-02-25:
Also, this is in the wrong place.   It's a GUI issue that requires better firmware to support the needs of the GUI.

Consider re-opening in the cleanflight-configurator issue tracker after searching for similar issues please.

closed 2016-02-25 by hydra

eddietheperson, 2016-02-25:
Thank you, will do.