cleanflight/cleanflight #1884: CC3D Sonar Unavailable

Quajetson, 2016-02-23:
I flashed CC3D with many versions of Cleanflight - latest as single image mode
using FTDI and STM FLASH LOADER in bootloader mode but when i TYPE CLI COMMAND
feature sonar, always getting response UNAVAILABLE but on gui can be enabled.
also not available sonar checkbox on SENSOR tab in gui.

RX_PPM -enabled
sonar connected as 4 & 5 to RC PORT

any solution?

jflyper, 2016-02-23:
Sonar seems to be disabled (not configured/compiled) on CC3D target by default for some unknown reason.

See src/main/target/CC3D/target.h, around line 123.

3djc, 2016-02-23:
Apparently,  it was removed as part of adding airport :https://github.com/cleanflight/cleanflight/commit/e778324c2d140c62e0bcd411850721799256be3b#diff-88749997b7c046162ab6f467491f98ff

Quajetson, 2016-02-23:
Which of the older version Cleanfligt I need to use sonar?

3djc, 2016-02-23:
I think it was introduced as part of 1.12, so 1.11 should do.

Quajetson, 2016-02-23:
OK i try tomorrow. Thanks for reply

Quajetson, 2016-02-24:
In 1.11 sonar works:) Thanks

closed 2016-02-24 by Quajetson