cleanflight/cleanflight #1885: Jeti duplex

divenet, 2016-02-23:
I am a fan of the work that went into Cleanflight
Here is a quick question, i have a Jeti 2.4 duplex that use EX Bus, any chance of seeing that being supported, as i am sure i am not the only one with love for an exquisitely made radio. currently works great in ppm but would love to step it up.

dkisselev, 2016-02-24:
Jeti EX has been implemented by another contributor and is just awaiting final review/merge, so you should see it in the next few releases: https://github.com/cleanflight/cleanflight/pull/1709

divenet, 2016-03-07:
God I am so looking forward to that,  have my cake and eat it too. Thank for all the hard work that goes into the marriage of the awesome Jeti and great CleanFlight there kid the SPRacing Pro F 3 mini, the best flight platform I have ever toyed with.. You are the best.

SnipeXR, 2016-03-08:
I have tried it in betaflight works great! Telemetry works too Voltage and Current!

divenet, 2016-03-08:
thank you thank you so much for the update, in you got any input, trick or learning curve you can share, ill bring it up to all my Jeti brothers here in Canada

SnipeXR, 2016-03-08:
Just make sure to install a resistor to brige Uart tx and rx together.  2.4k ohm 
See 
https://imgur.com/a/kGn0M
Enable telemetry and select exbus as serial provider on CLI and good to go.

divenet, 2016-03-09:
"Just make sure to install a resistor to brige Uart tx and rx together. 2.4k ohm
See
https://imgur.com/a/kGn0M
Enable telemetry and select exbus as serial provider on CLI and good to go. "

wait, do i see Grd and 5V. on pin 1 and 2 but signal on pin 6(4) with 2.4k bridge to pin 5 (3), did i get that right? and enable ex bus in serial provider? all of this will work on a naze 32? 

all this on port uart1?

and with any luck, same thing on a SPRacing F3 board?

you the Man thanks for your time

SnipeXR, 2016-03-09:
Yeah you got it except for its Uart 2 I had hooked up to. 
All works on Naze 32. Here's another pick from radio. 
https://imgur.com/a/boawA

SnipeXR, 2016-03-09:
I hope hydra merges the #1709 pull request. Works great.

divenet, 2016-03-09:
you are the man.

yup sorry, my bad, it would be Uart 2  at pin 3 and pin 4, guess i am going to bed late to give this a go.

so Uart 1 can remain free for usb or ODS when Armed.

i am so please to hear this, you have just made my day, now i feel good to have gone against the grain and got a jeti just because it is so well made.

i am indebted to your superior knowledge.

JL

marv-t, 2016-03-09:
Hi!
Currently, the following sensors are supported :

### exBusSensor_t jetiExSensors[] = {
    { "Cleanflight D1", "",     0,      0,             0 },                     // device description
    { "Voltage",        "V",    0,      EX_TYPE_14b,   DECIMAL_MASK(1) },
    { "Current",        "A",    0,      EX_TYPE_14b,   DECIMAL_MASK(2) },
    { "Altitude",       "m",    0,      EX_TYPE_14b,   DECIMAL_MASK(2) },
    { "Capacity",       "mAh",  0,      EX_TYPE_22b,   DECIMAL_MASK(0) }}; 

For implementation other values just ask.

divenet, 2016-03-09:
HI marv-t
you would make my system complete if there is any changes to have GPS coordinates?
i don't know if it possible but that would rock

JL

marv-t, 2016-03-10:
I do not have GPS, but have just ordered one for testing. However, it is possible in principle.

divenet, 2016-03-10:
> I do not have GPS, but have just ordered one for testing. However, it is possible in principle.

I have a Neo 8 and Neo 6 using Ublox provider on a Naze 32 and on my OSD, the data from the FC is to the minute accurate. Just a little hard of long to get a signal indoors as you would know for testing but outside it locks on as soon as you power up.

I would guess that you would test the location in the string of data and the length of the string to fit the Jeti protocol? Is that how it works in general and then translat to EX Bus before pumping the data into telemetry back to the RX?

Thank you so much marv-t for doing this for the Jeti Community but it kinda makes me feel like i had a hand in it if i were smarter.
JL

divenet, 2016-03-11:
thanks to all of you for the work done for the Jeti community .
JL aka Heliox

schaetj, 2016-03-13:
Thank you for implementing this.  What wattage for the resistor?  Is 1/8 watt sufficient?  5.0V / 2400ohm = 0.002083amps.  0.002083 ^ 2 * 2400 = 0.0104watts.  Well within 1/8 watt.

SnipeXR, 2016-03-13:
1/8 watt works that's what I use.

divenet, 2016-04-20:
Can one of you explain what is the purpose of the resistance and will it be the same implementation on other board Uart port? 
I mean this will work on a naze32 type flight controller but will it be the same setup  on a Seriously Pro Evo or the Kiss FC?

marv-t, 2016-04-21:
@divenet 
The resistor protects the IOs. It is transmitted and received via one line, so both ports are connected to each other (half duplex).
This works with any board that runs cleanflight, betaflight or raceflight. ;-)

divenet, 2016-04-21:
thanks marv-t

divenet, 2016-04-26:
Ok i am so sorry but i have to ask, in this sting of message, there is two values for the resister for the hardware setup. one place states 2.4k and another states 2.7k, i know its not much difference, ie 300 ohms. is one more accurate than the other for the application.  

Input Please @SnipeXR and @marv-t 

PS. for everybody's info, i will install this resister on a SPR f3 evo, ill be using a surface mount that fits just perfectly between Tx an Rx

SnipeXR, 2016-04-26:
Ethier one will work that you mentioned. There is not a tight tolarance needed it's just used to help protect the devices on a half duplex wire comms.

divenet, 2016-04-26:
> Ethier one will work that you mentioned. There is not a tight tolarance needed it's just used to help protect the devices on a half duplex wire comms. 

thanks @SnipeXR

marv-t, 2016-04-27:
@divenet 
I've used Mini Melf (MMA). Fits very well to the pins in 2,54 mm pitch.

[image]

divenet, 2016-04-27:
> I've used Mini Melf (MMA). Fits very well to the pins in 2,54 mm pitch.

Tanks @marv-t , very nice work and great idea

martinbudden, 2016-04-28:
@divenet , can you close this issue please?

divenet, 2016-04-28:
@martinbudden sure can, thanks to all.

closed 2016-04-28 by divenet