cleanflight/cleanflight #1889: TBS Gemini support (Colibri FC)

idefixsin, 2016-02-24:
Hi guys,

Just throwing something in the room. Let me know if this is nonsense,not possible, was discussed before and discarded.....

I'm using a fork of CF with a TBS Gemini (Colibri) and was wondering how large the differences to colibri race would be to support the gemini colibri FC through the main branch of CF.
The guys here have done an amazing job with CF 1.10.0 on the colibri (http://fpvlab.com/forums/showthread.php?42273-Time-for-Cleanflight-on-the-Gemini!/) but it looks like but continued support of future releases might not be guaranteed if it remains a fork. 

Any thoughts?

Thanks !

hydra, 2016-02-25:
If they raise a PR we can review and merge.

sukoshi1507, 2016-03-01:
Basically, everything is in place in this fork, which is the one i used to release the latest CleanFlight for the Gemini. 

https://github.com/sambas/cleanflight/ most of the work has been done by sambas in order to support the STM32F4xx with CleanFlight.

idefixsin, 2016-03-02:
As the current version of the code is based on CF 1.10.0, it was declined to be merged. 

Dominic, we would need to uplift the code to the current  master. Before we attempt that,  is there a chance that this will be merged in the first place? The TBS Gemini FC currently requires the Taulabs SW to flash the hex (no bootloader pins). I have no idea whether CF configurator coule be enabled to flash from there directly. Till this happens, the flashing part will likely remain a separate affair, similar to Sparky in the past.  You cool with that? 

Thanks

sukoshi1507, 2016-03-02:
@idefixsin : Let's work on the F4 portability in current master.

IdefixRC, 2016-03-20:
Hi guys,

I have created a new project on github to port the original code from Sambas and team to CF 1.10 release.
As I'm a bit rusty (very little programming in the last 15 years), I'm happy for any pointers and help you guys can give me.

Here it is so far:
Master = CF 1.10 Release: https://github.com/IdefixRC/cleanflight-R1.10/tree/master

Branch 1 (TBS-Code-CF-1.10-from-Sambas) =  Orginal code from Sambas and team - works with the colibri F4: https://github.com/IdefixRC/cleanflight-R1.10/tree/TBS-Code-CF-1.10-from-Sambas

Branch 2 (Colibri-Development) = CF 1.10 with the F4 code merged: https://github.com/IdefixRC/cleanflight-R1.10/tree/Colibri-Development

Branch 2 is the work in progress code.
So far I have tried to find all relevant files and code which relates to the Colibri (Revo and AnyFC ignored as well as all the ports not required for Colibri)

My plan is to get the CF 1.10 Colibri code working and after that uplift to CF 1.12, followed by CF current master in order to merge with the original CF master.

Major areas that I believe will require work are the serial and spi interfaces as the code used by the original team uses a different structure vs any of the CF releases I could find.

closed 2017-02-25 by hydra