cleanflight/cleanflight #1890: Why are master_t related header files are not included in config_master.h?

jflyper, 2016-02-25:
It may be a FAQ, but wouldn't it be nice to include all master_t related header files in the config_master.h so that every file that includes config_master.h do not have to include these header files individually?
Dependency handling?

I was playing around with a new config element defined in a new header file and found out that I have to visit and edit every file that includes config_master.h to include the new header file, and wondered why.

hydra, 2016-02-25:
so that when you open the file you see EXACTLY what the depenencies are.  This aids greatly with hunting down code smells, fixing dependency issues and helps make the design better - e.g. because of this there are now ideas on how to re-organize the configuration completely!  Once that is done liklely the profile/config stuff will go and the dependencies will be cleaned up not only in the production code but in the test code too.

closed 2016-02-25 by hydra

jflyper, 2016-02-25:
Thanks for the explanation!