cleanflight/cleanflight #1891: SPF3 / X4RSB - Flashed FC with cleanflight 1.12 issue

allgoodfun, 2016-02-26:
SPF3 / X4RSB - Flashed FC with cleanflight 1.12 from 1.10
Can't get transmitter to correspond correctly with receiver tab in CF!
tried all 3 different settings but they made very little difference!

Flashed back to CF 1.10 for now.

allgoodfun, 2016-02-26:
Current set up
sbus to IO2 / UART3
LED to IO1
telemetry to UART2
vbat connected
Buzzer Connected

AmiSMB, 2016-02-29:
I found that going from 1.10 or 1.11 to 1.12 all of the settings are changes as well as many defaults. You need to make sure that the settings are still set for serial to UART 3 but you also need to make sure that the following are set:
Feature RX_SERIAL is enabled
serialrx_provider is set to SBUS

allgoodfun, 2016-02-29:
perfect, thank you.
hadnt checked:
Feature RX_SERIAL is enabled
serialrx_provider is set to SBUS

martinbudden, 2016-03-01:
@allgoodfun , if that fixes the problem, can you close the issue? It's good to get issues closed as soon as they are resolved.

closed 2016-03-01 by allgoodfun