cleanflight/cleanflight #1892: SP F3 acro not sending vfas telemetry for open tx 2.1.7

jeffwhu, 2016-02-26:
On open tx 2.1.7 and sp f3 fc I cannot get the vfas to display when using "discover new sensors", just gives SWR,RSSI, rxbat and A2, have followed same procedure with an acro naze board and when armed I get a whole load of sensors, which includes VFAS, how can I get the sp f3 board to do the same??
Also, I have an x4r sb configured to sbus, connecting to tx pin on uart 2 to "S" pin on x4r I lose power to the fc board and receiver, disconnect the wires and it all reverts back to normal??? I don't get this at all?

AmiSMB, 2016-02-29:
The x4r sbus connection should not be connected to UART 2 as this is a 5v signal line when SBUS is 3.3v. UART 2 can be connected to the Smart Port for telemetry which is a 5v signal, but the sbus connection must be made to UART 3 or IO_2 CH3 connection

jeffwhu, 2016-02-29:
Thanks for the info, I think I have connected to IO2 pin 3 to sbus, sbus is on hart 3 to x4r, I followed "painless 360" tutorial on you tube.
I'll have a good look at it tomorrow and post what I find.
Thanks again for your help
Jeff

jeffwhu, 2016-03-02:
[image]
Hi 
as you can see I have got pin 3 on uart2 connected to smart port on x4r as you advised, have smart port enabled in ports settings in cleanflight and I have the same problem as I originally posted along with the power going. Excuse the soldering,just a temporary set up green wire from uart2soldered to yellow wire going to x4r smart port.
Any idea what is going on??
Thanks
Jeff

jeffwhu, 2016-03-02:
PS
All other wires not connected

jeffwhu, 2016-03-08:
I thought I would try and use an acro naze32 rev6 and have exactly same problem as the original question,
HELP NEEDED PLEASE

closed 2016-03-17 by jeffwhu