cleanflight/cleanflight #1893: Allow Default Flight Mode To Be Set by CLI (and GUI)

OleEngr, 2016-02-26:
Having the default flight mode set to ACRO says something about the importance of racing to the developers.  However, there are folks that might never use ACRO, and would prefer a default that doesn't require further tinkering in the MODE tab (maybe for a camera platform).  

There is no reason to change the current default, but a CLI option should be a painless change.  Better  yet would be a GUI selection, so those folks that never want to see the CLI stuff can painlessly get their choice registered.

3djc, 2016-02-26:
Can't you do something similar by setting your flight mode of choice to the full extent of an aux channel,  so that it is simply always active?

Arakon, 2016-02-26:
Exactly. And default mode is not Acro. Default mode is "no mode", which happens to equal Acro.

OleEngr, 2016-02-26:
@ 3djc  Of course, and that's exactly what I did, when I did want any other modes available . 

The point of this is that you shouldn't have to jump through hoops if you all you want a simple, single mode selection, and nothing else.

joshuabardwell, 2016-02-29:
Acro is the default mode because gyro-based control is the foundation on which every single other flight mode is based. The gyro is always active, no matter what other flight modes there are. It has nothing to do with an affinity for racing.

The ability to select other default flight modes already exists, via the Modes tab. I don't understand why you think that creating an always-on mode is "jumping through hoops" but configuring a CLI option is not. Adding duplicate, conflicting functionality would be confusing and counter-productive.

proggod, 2016-02-29:
This is really more of a configurator issue, and something like I suggested in this issue would be a good place for it:

https://github.com/cleanflight/cleanflight-configurator/issues/349

I'll be glad to write the code, just need Hydra or someone to talk to me and sign off on it.. :)

OleEngr, 2016-03-01:
My motivation for opening this issue was to simplify the flight mode selection for new users, that might only be interested in Angle mode.  I understand how the logic works, and how it might have originated, but I think it is unnecessarily complex to define the default flight mode as: that which results when Not-Angle and Not-Horizon modes are selected.  More than complex, I think it is crude to expect someone that only wants Angle mode to have to set up a switch that they never intend to use, implemented with a resource they may not have on their transmitter, e.g. Devo 7e.  My suggestion to add it to CLI as an interim was BAD idea;  It belongs in the GUI where it has clear visibility.  If there is anyway to give a new user a case of overwhelming-itis, it is to require any additional CLI settings for typical use.
The addition of the Failsafe capability to the GUI is a perfect example of a needed addition to the GUI.  AFAIK all of the parameters were available in the CLI, but the ease of putting it together was not even close.

proggod, 2016-03-01:
This is really why I think we need a setup wizard, when first running it, it could ask you how you want to use the drone, and set it accordingly.  So new users would be set up with horizon mode always on with lower roll rate, more advanced users set up with multiple modes and higher roll rates, etc

closed 2017-02-25 by hydra