cleanflight/cleanflight #1895: naze32 v1.12 , gps 3D fix + arming problem

emski0, 2016-02-26:
Once the GPS find a 3D fix, the naze32 stops arming.
Reverted back to 1.11, no problem there, once  it get 3d Fix , i can still arm the naze32

OleEngr, 2016-02-26:
That sounds exactly like the problem I ran into in V12 released.  I didn't equate it to the GPS, but I knew it was something in my configuration, and that if I deleted enough from the config it eventually armed.  I quit trying to figure it out and went back to V11.  The GPS modes in V11 don't work correctly, but maybe that is a PID problem in my config.

3djc, 2016-02-27:
1.12 is very demanding for F1 processor with the default settings. It does work, but if you add enough work on top of default config, it will break. I think this is what is happening here. Try setting in CLI : 
set gyro_sync_denom = 2

OleEngr, 2016-02-27:
Setting gyro_sync_denom = 2 allowed the quad to arm again.  Thanks for the tip.  There were a few times when it did arm with V12 and the GPS enabled, but maybe there wasn't a fix to raise the processor load the point where it wouldn't arm.  Now, if I could just get the GPS and ALT to work as well as they did in V12-RC1.

borisbstyle, 2016-03-02:
There should be some kind of auto configuration when some features enabled

yedey, 2016-03-04:
I had the same issue, but I couldn't arm even if gps wasn't fixed. The "set gyro_sync_denom = 2" workaround fixed it. I agree with borisbstyle, this should be configured automatically. It took me a whole day to realize that it wasn't a fault of my cleanflight/GPS/TX setup.

hydra, 2016-03-09:
fixed in ee6b0df418498d9252b2a507f93f7aa1430dcde6

gyro sync denom forced to 2 when gyro sync and gp enabled at the same time.

dyter, 2016-03-11:
check your cpu load with status command

martinbudden, 2016-05-30:
@hydra, I think this can be closed now.

closed 2017-02-25 by hydra