cleanflight/cleanflight #1898: NAZE32 REV6 Telemetry

plogibear, 2016-02-27:
Hey guys, got a quick question with the NAZE32 REV6 and Cleanflight v1.11.0. First of all It does seem to work and fly well BUT when I set it up for Telemetry some interesting things happen..

I got the board to output telemetry via softserial1(Pin6) to the D4R-ii/Taranis and I'm getting telemetry BUT ONLY WHEN ITS ARMED. Oh and I had to invert the signal ofcourse to get telemetry otherwise i dont.

As of now nothing extra on the board except that..

What am I missing here? (AFAIK with Rev5 you can get telemetry via Softserial regardless when its armed or not. Well that was what I wanted on the rev6).

By the way I also tried 1.11 or 1.12 but cant seem to arm the board nor get telemetry.

What do you guys suggest?

SteveAmor, 2016-02-27:
Forums are for support. Please can you close this issue as it diverts developers attention. Thanks.

plogibear, 2016-02-28:
Hi Steve,

I understand thats why Im reporting that there is an issue within cleanflight with the rev6 board.

plogibear, 2016-02-29:
Got it fixed and figured out. I forgot to uncheck MSP at softserial1.

closed 2016-02-29 by plogibear