cleanflight/cleanflight #1899: Motor ouput pins changed in master?

marv-t, 2016-02-27:
Here is the output form M1-M6. The M3 changed to M5 and M4 to M6

[image]

In 1.12.0 NAZE 2016-2-20 14:55(stable) is it correct.

[image]

marv-t, 2016-02-27:
It seems Y6 is also not working...

[image]

tricopterY, 2016-02-28:
In CLI do a dump and see if  'feature SERVO_TILT'  is disabled i.e displayed with a minus sign:
feature -SERVO_TILT

marv-t, 2016-02-29:
@tricopterY The feature SERVO_TILT is disabled. Only in master is the pin mapping wrong.

SteveAmor, 2016-02-29:
@marv-t can you do a git bisect to see which commit introduced this issue?

marv-t, 2016-03-04:
Is solved by ledvinap.

closed 2016-03-04 by marv-t