cleanflight/cleanflight #1903: SP-F3 Acro (all versions of SP-F3 firmware) not finding external mag BMP180

CaLviNx, 2016-02-29:
When using the genuine acro board and the latest SP-F3 hex in the latest clean flight an external baro (BMP180) is not being recognized, tried a few different boards to make sure its not the external board that is faulty. Attempted to manually set the board type using the CLI but still no joy.

The bmp180 boards not being seen by the SP-F3 board using the latest hex are automatically found in cleanflight using both a flip32+ and a Naze32 Rev 5 FC's.  Tellingly when the SP-F3 is flashed with the correct version of the "iNav" firmware the external BMP180 is recognized instantly.

closed 2017-02-25 by hydra

matevz25, 2017-08-28:
Hi, Iam new at this... how do I download modified fw?
Thank you!

mikeller, 2017-08-28:
@matevz25: This has long been released in an official version of cleanflight. Just use the cleanflight configurator to update your board to the latest firmware.